erdl/lonoa #1: Create Unix timestamp conversion function to datetime in CSV files

ryantanaka, 2017-09-14:
https://gist.github.com/ryantanaka/33ea250414d23f27787ec0eed93c9b4c

To do: 

- need to change datetime to strictly use UTC-10 when doing conversion.

### Motivation
- running this function on the output of 12 python files from scrape-util to convert unix time stamps in csv file to datetime format. the notebook that generates this output will be recreated to no longer use the 12 py files, and this function may be used directly.

carlosviansi, 2017-09-14:
I will take it from here, once the notebooks are created for [WebCTRL](https://github.com/erdl/scrape-util/issues/5) and [egauge](https://github.com/erdl/scrape-util/issues/1) I will see how to best use this.

carlosviansi, 2017-09-23:
@ryantanaka We will be using arrow from now on for our sanity: http://arrow.readthedocs.io/en/latest/ 

I used on the [eGauge Notebook](https://github.com/erdl/sensors/blob/a5e4bcebb632b0cb454cb26bcf5559cc8b828f34/egauge/Requesting%20raw%20data%20from%20eGauge%20API.ipynb).

I guess the syntax is as clear as it can get.

closed 2017-09-23 by carlosviansi