erdl/lonoa #5: Data for DHHL Reports from KHIKAPOL19, KHIKAPOL23, and KHIKAPOL8 in Weather Underground

carlosparadis, 2017-11-07:
[image]

## What Stations

Stations are highlighted in the image above. Associated links to the stations in Weather Underground website:

 * https://www.wunderground.com/personal-weather-station/dashboard?ID=KHIKAPOL19
 * https://www.wunderground.com/personal-weather-station/dashboard?ID=KHIKAPOL23
 * https://www.wunderground.com/personal-weather-station/dashboard?ID=KHIKAPOL8

## Time Window

1 Year for each station aligned. 

## Script

To be clear, this is not done through scrape-utils, but by the [Weather Underground Notebook](https://github.com/erdl/sensors/tree/master/weather_underground) in this repository. 

## Storage

Data will be stored as separated tables in the database, but with their timestamp converted to HST. Timestamp field will **not** contain a timezone column, as it breaks Tableau (!). 

To guarantee consistency in the timezone, data will be requested in UTC instead of "guess my local time", explicitly converted to HST by the Notebook, and then stripped of its time zone.

carlosparadis, 2017-11-08:
@eileenpeppard I have the notebooks done to download the raw data and parse into a single .csv on **HST** timezone. 

I can only download 1 year of data per day, so I used Tuesday to download KHIKAPOL19, and am using Wensday already to download KHIKAPOL8. The time window is 20161106-20171106. 

The data for both stations are on the Google Drive: CSBD/DHHL/Kat data from Eileen/weather_underground. 

@kathrynparadis told me she is practicing adding data to the database "just for fun", and you want this data in the database, so I guess this is a perfect fit to be deferred to her. 

Note the .csv already has the timestamp in HST in a naive format (i.e. it doesn't include the timestamp that Tableau can't deal with). So once this table is added, it should work immediately in Tableau. 

**Important**: Do note the number of sensors varies per station and I have no control over that. Also, I already observed some sensors have problems with wrong readings, I am leaving this up to you. Kat said she would be in on Thursday, so you will be getting your data much earlier than next week where you wanted. 

If you are curious about the notebooks please see these 2 links for the one that downloads the raw data, and the other that parses it into a .csv:

 * https://github.com/erdl/sensors/blob/master/weather_underground/Weather%20Underground%20API%20json%20Data%20Acquisition.ipynb
 * http://nbviewer.jupyter.org/github/erdl/sensors/blob/091a132ebd8f2da6e9de1444de0b7f9f68709b75/weather_underground/Weather%20Underground%20json%20to%20csv%20Data%20Parser.ipynb

Note the second notebook plots the number of days missing data just after you run it, so you have an idea if the data is even worth putting on the database (e.g. from when I ran the parser for KHIKAPOL19):

[image]

I wrote enough explanation so that even Pipi could run the notebook. Please let me know how it goes on Thursday so I can close this issue afterward.

ryantanaka, 2017-11-09:
[image]
KHIKAPOL23 station missing days.

kathrynparadis, 2017-11-12:
Carlos ran a script to create code for individual tables for each station. Once the code was created, we corrected the formatting, and created each table using the 3 following pieces of code: 
```
CREATE TABLE KHIKAPOL8 (
"UV" float8,
dewpti float8,
dewptm float8,
heatindexi float8,
heatindexm float8,
hum float8,
precip_ratei float8,
precip_ratem float8,
precip_totali float8,
precip_totalm float8,
pressurei float8,
pressurem float8,
solarradiation float8,
tempi float8,
tempm float8,
wdird float8,
wdire VARCHAR(255),
wgusti float8,
wgustm float8,
windchilli float8,
windchillm float8,
wspdi float8,
wspdm float8,
datetime TIMESTAMP WITHOUT TIME ZONE
);
```

```
CREATE TABLE KHIKAPOL19 (
"UV" float8,
dewpti float8,
dewptm float8,
heatindexi float8,
heatindexm float8,
hum float8,
precip_ratei float8,
precip_ratem float8,
precip_totali float8,
precip_totalm float8,
pressurei float8,
pressurem float8,
solarradiation float8,
tempi float8,
tempm float8,
wdird float8,
wdire VARCHAR(255),
wgusti float8,
wgustm float8,
windchilli float8,
windchillm float8,
wspdi float8,
wspdm float8,
datetime TIMESTAMP WITHOUT TIME ZONE
);
```

```
CREATE TABLE KHIKAPOL23 (
"UV" float8,
dewpti float8,
dewptm float8,
heatindexi float8,
heatindexm float8,
hum float8,
precip_ratei float8,
precip_ratem float8,
precip_totali float8,
precip_totalm float8,
pressurei float8,
pressurem float8,
solarradiation float8,
tempi float8,
tempm float8,
wdird float8,
wdire VARCHAR(255),
wgusti float8,
wgustm float8,
windchilli float8,
windchillm float8,
wspdi float8,
wspdm float8,
datetime TIMESTAMP WITHOUT TIME ZONE
);
```
Once the tables were created, we took the csv files that Carlos extracted from the website, and uploaded them to their corresponding table.

eileenpeppard, 2017-11-15:
thanks for the info on weather stations!

carlosparadis, 2017-11-15:
@eileenpeppard Glad this issue is finally solved after all these months (and the notebook to get it done, albeit it cost an entire day of my life on it). 

## Why this issue is still open

This issue will remain open until I have time to move the table schema Kat just posted in ERDL/databases. 

I always prefer having the actual schema that was used to create than the one the database toss at you (this is especially true for when we use the `serial` alias since it becomes a bunch of other things after it is used in the database).

carlosparadis, 2018-04-30:
## Update 

Given much has been changed in the schema, and this data was never really used other than stored in the database, I am closing the issue. For reference, the original intention of this data was to help supply input to BIOP / Energy Plus, but Riley and Arch ERDL et al. decided to use: 

https://energyplus.net/weather-location/north_and_central_america_wmo_region_4/USA/HI/USA_HI_Barbers.Point.NAS.911780_TMY3

See thermal comfort's issue on data from the link above for more details:

https://github.com/erdl/thermal_comfort/issues/17

closed 2018-04-30 by carlosparadis