erdl/lonoa #10: Document project requirements

ryantanaka, 2018-02-05:
# WIP

# Requirements for Sensors Project

The requirements for the sensors project will be documented here. This will serve as a proposal draft for the project before any coding begins.

## 1. Project configuration files must be stored in the database.
## 2. Error logging must be provided for each project. 
## 3. Each project needs to be run as its own program.

ryantanaka, 2018-02-26:
meeting w/ Eileen 2/26/18 notes:
- discussed database schema 
    - what we should have (https://github.com/erdl/database/blob/master/README.md)
     vs.
    - what Eileen is able to use
    - possible tradeoffs 

- webctrl is fixed in the frog buildings and will not move
    - don't have to worry about knowing where a webctrl sensor was in a prior project when looking
at the reading table

- egauges
    - can be moved
    - can yield different readings depending on how they are set up

TODO: come up with a simple schema incorporating tradeoffs from the way we should do it versus what Eileen is able to use easily and report back to discuss

carlosparadis, 2018-04-30:
We can re-open this issue if we ever get back to this.

closed 2018-04-30 by carlosparadis