erdl/lonoa #11: Create minimal code to log VOC and Carbon Dioxide Sensor with Timestamps for Raspberry PI

carlosparadis, 2018-03-02:
Let's try to start small with these sensor projects so everyone can learn how the data and code will look alike for a future scrape-util utility. I believe we can also explore a lot of details for this particular small project which can serve as prototype documentation for following and more complex ones. 

## Goal

The goal of this issue is to save to the rhasberry pi a .csv file of the following format: 

| timestamp | reading1 | reading2 | reading3 | ... |
| ----- | ---- | ---- | ---- | ---|

For the sensor: 

[Adafruit CCS811 Air Quality Sensor I2C Breakout - VOC & eCO2 - 3v3 & 5v -Arduino](https://www.ebay.com/itm/Adafruit-CCS811-Air-Quality-Sensor-I2C-Breakout-VOC-eCO2-3v3-5v-Arduino/172833181875?hash=item283da910b3:g:XRAAAOSw6WVZnN7v)

[image]

and the RTC: 

[DS3231 AT24C32 IIC precision Real time clock RTC memory module](https://www.ebay.com/itm/DS3231-AT24C32-IIC-precision-Real-time-clock-RTC-memory-module-For-Arduino/201478295970?hash=item2ee90ad1a2:g:UA8AAOSwHQ9WYHza)

[image]

Which Eileen is interested and you already purchased. 

## Code 

Feel reminders here: 

 * Make it clear what is code associated with each component. Here you have 2 parts, but one of them has 2 sensors. It is important we can read your code and make sense which one of them it is interacting with (and preferable why). 
 * The rest of the group doesn't have much background on where to get the code to deal with the sensor. Make sure to document where you got the code, any examples that help you get through, or documentation of the code the sensor may have provided. 
 * If you feel a tiny mistake may lead to fry the entire sensor on the code, make sure to put a lot of red flags. It may not be as obvious to us.  
 * Meaningful variables in the code here will go a long way, especially if you are using hexacode or assembly to specify some pin. I have seen from easy to read to completely gibberish code out there for sensors on Arduino and Pi. **Avoid acronyms and abbreviations, the rest of the group will likely not recognize jargon!**

## Documentation

3 Things: 

   - [ ] **Text**: You don't need to write a book on it, but try to write enough someone from your major can at least replicate, as you saw on the [Air Quality Station](https://www.rigacci.org/wiki/doku.php/doc/appunti/hardware/raspberrypi_air) or Sparkfun's tutorial. Whichever you feel it is more instructive. 
  -  [ ] **Diagram**: http://fritzing.org/home/ 
  -  [ ] **Photos**: A few photos of the actual thing, especially on spots that may be tricky 

If you know any other tool than fritzing, feel free to suggest.

carlosparadis, 2018-03-09:
@jygh98 mentioned there is a `cold start` problem, where the first 5 readings may be biased. We need to check with @eileenpeppard wether she prefers to have biased data in the dataset for post-processing, or if we should go with some rule of thumb to always filter the first few rows when it starts in the code. 

Thoughts?

eileenpeppard, 2018-03-09:
Hi, guys. It would be preferable if it was filtered before storing data...  Hey, I found GitHub! :)

carlosparadis, 2018-03-10:
@jygh98 you heard the boss! Here is what I suggest we do: Try to run the code in a few different places to see if the environment has any impact on the decision on how many measurements we should not store on turning on. 

**Keep the data** you generated in these different places **stored** in your computer (as in a "experimental dataset") so it is reproducible your analysis of why X rows would be the safest to be discarded. 

**Maybe doing this experiment is better after you have the entire air quality station going, or at least the environment sensors so we can keep track also if the environment was different.**

Thoughts?

closed 2018-04-30 by carlosparadis