erdl/lonoa #17: Raspberry PI 2 runs too slow on most recent image

carlosparadis, 2018-04-30:
@jygh98 the PI 3 B+ you bought seems to work fine, but on the old one, using NOOBS v2.81's Rasbian 9 Stretch is slow (holding an on the terminal make it freeze sometimes and also lag while dragging the mouse.  Running updates also lag it to the point of freezing and requiring a reboot. 

Try to find out a solution to this. We may need to roll back to an older system version of Rasbian. 

See for example: https://www.raspberrypi.org/forums/viewtopic.php?t=208595 

For related discussion. Also, see if you can find anything about the end of life support for Jessie, in case you think installing that would be a good idea, and a way for us to obtain it, if possible via NOOBS to keep the setup simple.

carlosparadis, 2018-04-30:
For reference, 

https://wiki.debian.org/LTS 

Although this may not reflect the Rasbian variant. Some say the Rasbian variant of Jessie may hit end of life on May 2018:

https://www.raspberrypi.org/forums/viewtopic.php?t=182288 

If that is the case, we are semi-doomed lol.

carlosparadis, 2018-05-01:
I was misled. Jessie also runs slow, but I found the problem: The sensors. After I unplugged them (with it powered off), it started running smoothly again. I didn't want to try my luck on the right GPIO, so I didn't try to plug them back in. 

@jygh98 Any idea, if the fact gpio library is not installed, could make the thing slow? I am surprised just the fact they are connected without any script running is slowing it down. 

I am going to re-install Stretch and see if it runs fine without the sensors.

carlosparadis, 2018-05-01:
Can confirm on Stretch the pi runs just fine without the sensors plugged. Let me know what you think about y question on gpio library. Still find weird it lags it down without running anything..

jygh98, 2018-05-01:
Hmm interesting, off the top of my head i cant imagine why that would happen.  But if that is the case then i guess when i start repopulating back the sensors on the raspberry pi maybe i can experiment what caused this slow down.

carlosparadis, 2018-05-01:
@jygh98 Alright sounds good. Once @ryantanaka put the pi 3 B+ on your desk on a port in erdl, I should be able to sudo and install what you need. 

@kathrynparadis should stop to just drop the spaghetti of sensors (I did not remove any cable from the breadboard or modified. It is "as-is" but does double check in case something fell off. But this issue should not be your top priority (the priority is getting the EMP code from the new sensor going). 

However, Pi 2 will stay in my room since I didn't have time to configure yet. 

I think both of them will only come in the afternoon, so if you could also come later it would work best. 

In the meantime, I suggest you update the wiki with the remaining photos and/or check with ryan how to host the images on the wiki if you don't have anything else to do :)

carlosparadis, 2018-05-02:
@jygh98 I forgot this was already open. Just update here if you find anything that justifies why previously it did not lag with all the sensor and now does.

jygh98, 2018-08-21:
I was not able to find any specific reason why the old pi was running slow at the time, but as of now it appears to be running fine.

closed 2018-08-21 by jygh98