erdl/lonoa #21: Identify candidate sensor that is sufficiently accurate to HOBO for Raspberry PI

carlosparadis, 2018-05-02:
http://www.onsetcomp.com/products/data-loggers/u12-012 

@jygh98 Make sure to talk with @eileenpeppard about how precise (how many decimals), and accurate (how off it is from the "true measurement -- like 2 degrees off) she needs, so we can identify a sensor that meets the requirement while being inexpensive. 

If we can nail this, then @eileenpeppard can stop making roundtrips to collect hobo data moving forward (data would be sent directly to our database, similar to egauge), and we would cut down on hobo cost.

eileenpeppard, 2018-05-02:
@jygh98 go to the Onset website and check on the precision and accuracy of the Hobo U12 and the ZW system of hobos.

jygh98, 2018-05-03:
So i [made a spreadsheet](https://docs.google.com/spreadsheets/d/1G8zX17R3gmrRBiqp15pooMiErAD5dqjbZoV3bfFE7H0/edit?usp=sharing) comparing the HOBO U12 and ZW series sensors to the various sensors i found on Adafruit.  

To summarize, the U12 is capable of measuring temperature, RH, and luminosity, while the ZW measure only temperature and RH.  The Si7021($6.95) and SHT31-D($13.95) both measure temperature and RH that matches the range and accuracy of the U12.  Although the SHT31-D is costs nearly double of the Si7021, it measures slightly more accurate than the Si7021.  For measuring luminosity, there is not a clear way of measuring the accuracy due to how a sensor responds to a certain wavelength of light.  Because of this, the only way of determining the accuracy is by testing ourselves.

carlosparadis, 2018-05-19:
@jygh98 You need to keep tabs on the closed issues or I will go mad with so many open to keep eyes on :^) I am closing this one since we not only solved this, but you already bought and even coded it!