erdl/lonoa #22: Check if PIs running 24/7 would be better of with heatsinks

carlosparadis, 2018-05-02:
Just for the peace of mind!

jygh98, 2018-05-03:
In terms of reliability, from threads i been viewing people have responded saying that they were able to run the pi headless (meaning no monitor, mouse, keyboard) for months on end, even years without problems.  So the raspberry pi is capable of running 24/7 for long periods of time.

A heatsink is not required since the pi does not consume much power and therefore doesnt create much heat.  However it might be worth investing, especially since they are so cheap in bulk, since we are planning to run these 24/7 for long periods of time.  I doubt that it will make much of a difference, but it wouldnt hurt to have one anyway.

carlosparadis, 2018-05-03:
@jygh98 link some candidate parts here when you get the chance then :^)

jygh98, 2018-05-03:
[For one set its less than $1](https://www.ebay.com/itm/1-Set-3pcs-Aluminum-Heatsink-Cooler-Adhesive-Kit-for-Cooling-Raspberry-Pi-New/302231804440?hash=item465e6b5a18:g:AZkAAOSwCU1YqiFo)

[For ten sets i found this for $3](https://www.ebay.com/itm/10Set-30PCS-Aluminum-Heatsink-Cooler-Kit-For-Cooling-Raspberry-Pi/231908376753?hash=item35fed100b1:g:qbYAAOSwiylXCUV9)

Note that each set includes 3 pieces, one for the main cpu and the two smaller pieces are for other IC

carlosparadis, 2018-05-03:
I don't see why not get 10 sets for additional $2! Just send that one to Eileen then when we figure out the other parts. May also want to check with her if we should get a battery for the second pi when we buy the next set of parts.

carlosparadis, 2018-05-19:
@jygh98 Thanks, I believe this closes this issue.

closed 2018-05-19 by carlosparadis