erdl/lonoa #25: MATT

ryantanaka, 2018-05-04:
**First Day: May 11, 2018**

@ryantanaka, record screencast of survey_display and survey_admin for matt. 
- possibly use slides 
- make this comprehensive
- tag timestamps for topics 

starting task_1 (missing data training):
- use egauge notebook to get data for a given time period
- download the data, transform it into a csv, and send it to us 
- deliverable: csv file (run it yourself so you can just diff it)

starting task_2 (python script):
- turn the python notebook into a script and take the dates as args
- should produce the same output as the notebook

starting task_3 (timestamp):
- augment the script he previously wrote to write a txt file to disk that contains the state timestamp so that it can be re-read in by the program when it runs again
- note: not just latest timestamp, APPEND timestamps

starting task_4 (optional if he doesnt have virtual box and ubuntu):
- run some crontabs

starting task_5 (vagrant):
- teach him how to use the vagrant box 

starting task_6 deploy postgres on vagrant or virtual box
- now have timestamp stored on the database 
- should be something like `| sensor id | timestamp | event type (timeout, success, etc)|`

starting task_7 
- make a table in local database with csbcd configs so that it can be read (only need IDS don't translate this thing)
- using as "input": scrape-util config and eileen's sensor info table

carlosparadis, 2018-05-19:
@ryantanaka @matthew-schultz I formatted the body of the first issue comment above to make it easier to keep track of progress. @ryantanaka Please check the boxes as he progresses.

carlosparadis, 2019-01-08:
Since the effort for egauge, webctrl and hobo is done, and we are moving to refinements, I am closing this.

closed 2019-01-08 by carlosparadis