erdl/lonoa #28: Setup Sensors to Collect Data every 15 minutes

jygh98, 2018-05-12:
The data being stored should be an average of the 15 minute interval.

carlosparadis, 2018-05-19:
You forgot to mention this issue on the commits you made in various sensors for this (it is ok to put to pound signs with their commit numbers at the start if that happens to be the case). Could you put here the SHAs of the commits you did that so we can keep track?

carlosparadis, 2018-05-19:
For the record, refer to #36 on the rationale behind 15 minutes.

carlosparadis, 2018-06-02:
@ryantanaka Do you know if @jygh98 finished this before he left? We should touch basis on this so I can get some data via ssh.

ryantanaka, 2018-06-05:
@carlosparadis I believe so. The code is running on a cronjob on the raspberry pis atm. Looking at BME sensor code running on the 9S, it appears an average is being taken every 5 minutes instead of 15. 

https://github.com/erdl/sensors/blob/0b0f1f1aa5833cc9b04dd25846de55d3adf6dcd7/raspberry_pi/bme280/BME280_logger.py#L10-L11

https://github.com/erdl/sensors/blob/0b0f1f1aa5833cc9b04dd25846de55d3adf6dcd7/raspberry_pi/bme280/BME280_logger.py#L30-L54

The code is going to grab data, sleep for a second over and over 300 times (5 minutes) then take the average of those data points so I'm not sure where the 15 minutes came from.

#36 was opened 22 days ago while this issue (#28) was opened 24 days ago so based on the code and what was discussed in #36, I don't know if this issue is just outdated. 

As for data, it's there being collected. You just need to download the csv. When is convenient for you to go over this stuff?

carlosparadis, 2018-06-05:
That's a hard one :) Is your current schedule on google sites? I can try to accommodate around your time.

ryantanaka, 2018-06-12:
@carlosparadis  Haha, I'm in on Tuesday/Thursdays usually from 8:30 - about 11:30 Hawaii time. But for the time being, I put together a short [wiki](https://github.com/erdl/sensors/wiki/Raspberry-Pi:-Getting-Data).

jygh98, 2018-08-21:
Yes this issue has been finished, however i think the priority of this issue has been mostly shifted to #29.

closed 2018-08-21 by jygh98

jygh98, 2018-08-21:
We agreed 5 min resolution for all sensors.