erdl/lonoa #29: Setup a Crontab for the sensors

jygh98, 2018-05-12:


jygh98, 2018-05-15:
@ryantanaka

carlosparadis, 2018-05-17:
@erdl/dev Should we defer this to after @matthew-schultz is done with the main sensor scripts?

jygh98, 2018-05-23:
Currently i have set up user specific (jason) crontab on both 2B and 9S, where both crontabs are setup similarly with the only difference being the specific sensor script it is trying to run.  In the user specific crontab file, i have the crontab currently running every minute to execute a shell script in each of the sensor folders.

[image]

The shell script follows a basic template of checking if an existing process is running, if not it will execute the logger script.

jygh98, 2018-08-21:
The crontab has been set up on both raspberry pi as an user specific job (jason) and has been able to keep the logging script active.

As a reminder, each sensor folder contains the logging program `<sensor>_logger.py` and a shell script `<sensor>_check.sh` that checks for only one instance of the logging program is being run at all times.  This shell script is the one that is placed in the user specfic crontab, not the logging program.

closed 2018-08-21 by jygh98