erdl/lonoa #30: Setup a Database for the raspberry pi

jygh98, 2018-05-12:


carlosparadis, 2018-05-17:
@jygh98: Since @matthew-schultz will get around creating a database for the other sensors on task 6, and you have as priority having everything running first, I am re-assigning this to him.

@matthew-schultz You can safely ignore this issue until you are done with the entire issue #25. Ask me or @ryantanaka if you need more context about this out of curiosity, but does not interfere with your current work.

ryantanaka, 2018-05-21:
@carlosparadis, just want to confirm where this will be (eg ERDL desktop, IT-VM, the linux box that matt is currently using, etc.) 

I'm guessing we will test this locally, THEN the database should eventually be set up in the IT-VM once all the kinks have been ironed out in terms of the pi's working properly.

carlosparadis, 2018-05-22:
@ryantanaka yes, all the dev for now will go on the local server in ERDL. We will discuss moving all this effort to production later on and open a dedicated issue for "migration".

carlosparadis, 2019-05-21:
I'm closing this, as at the time replace Scrape-utils was not on the table. If the need arises, we should open a new issue to integrate to this Github project.

closed 2019-05-21 by carlosparadis