erdl/lonoa #36: Check with Eileen about sampling rate for the sensors

jygh98, 2018-05-15:
Need to figure out the measuring interval for the data we are collecting and also setting up an averaging function so that the measurements we are collecting are more accurate.

jygh98, 2018-05-16:
How often should it store data? 5 minutes
At what interval (granularity) is it sampling data? every second if possible

Because there is going to be a time interval in how often data is stored, Eileen would want us to do a trailing timestamp, where the timestamp in each row of data represents the time data was stored, i.e after 5 minutes has passed, it would take the time then.

carlosparadis, 2018-05-17:
@jygh98 Post here what you found out before closing the issue :)

closed 2018-05-19 by carlosparadis