erdl/lonoa #37: turn egague notebook into a script

ryantanaka, 2018-05-17:
Use the egauge notebook to create a script that obtains data from an egauge sensor, then writes that data to disk as a csv (append to the end of the file).

function arguments:

- unix start time
- unix end time
- egauge id
- path to file

You can post the code here when you're done and we will take a look at it.

matthew-schultz, 2018-05-17:
next tasks:
* check if request was successful before writing output to file
* figure out how to access header object in pandas dataframe object in python3

dataframe will eventually be inserted into database

matthew-schultz, 2018-05-18:
```
'''
Arguments: unix start timestamp, unix end timestamp, the egauge id, and output file path
example command: python3 egauge_api.py 1526432400 1526443200 725 ~/Projects/egauge/output.csv

This script will use the egauge api to get the data as a csv from the interval of time between the two timestamps and append that csv data (without the headers) to the output file path
'''

from io import StringIO

import os
import pandas as pd
import requests
import sys
import traceback

def main():
    if(len(sys.argv) >= 5):
        request_dict = get_args()
        host = set_host(request_dict)
        time_window = {'t': request_dict['start_timestamp'], 'f': request_dict['end_timestamp']}
        try:
            r = requests.get(host,params=time_window)
            if(r.status_code == requests.codes.ok):
                print('Request was successful', str(r))
                df = pd.read_csv(StringIO(r.text))
                '''with open(request_dict['csv_output_path'], 'a' ) as f:
                    #set header=False if we don't want to append header
                    #set index=False to prevent header prepending a comma
                    df.to_csv(f, sep=',', header=False, index=False)'''
                #print(str(df.))
                with pd.option_context('display.max_rows', None, 'display.max_columns', 4):
                    print(df)
                df_columns = get_header_string(df.columns.values)
            else:
                r.raise_for_status()
        except requests.exceptions.HTTPError as e:
            print(e)
    else:
        print('Invalid number of arguments. Run with example format: ')
        print('python3 egauge_api.py 1526548513 1526549515 34111 ~/Projects/egauge/output.csv')

#get the four arguments from sys.argv and return them as a dictionary
def get_args():
    request_dict = {}
    request_dict['start_timestamp'] = sys.argv[1]
    request_dict['end_timestamp'] = sys.argv[2]
    request_dict['sensor_id'] = sys.argv[3]
    request_dict['csv_output_path'] = sys.argv[4]
    return request_dict

def print_args():
    for arg in sys.argv:
        print ('arg is ' + arg)

def get_header_string(values):
    print('columns are: ', values)
    #print('columns are: ')
    #df_columns = df.columns.values[0] + ','
    #df_enum =
    #for df_column in df.columns.values:
        #df_columns = df_column + ','
    #print(df_columns)

# set host for egauge api get request
def set_host(request_dict):
    minutes = 'm'
    output_csv = 'c'
    delta_compression = 'C'
    host = 'http://egauge{}.egaug.es/cgi-bin/egauge-show?'
    host = host.format(request_dict['sensor_id']) + '&' \
    + minutes + '&' \
    + output_csv + '&' \
    + delta_compression
    return host

if __name__ == "__main__":
    main()

```

carlosparadis, 2018-05-18:
@matthew-schultz if you type `python` after ''' it will syntax color the code block for you (I edited your post to add that, try to edit your post so you can see how it looks like :)

matthew-schultz, 2018-05-23:
made a minor typo correction in the issue name ('gague' to 'gauge')

matthew-schultz, 2018-05-23:
I forgot to to add #37 issue marker to the commit message when I merged the latest updates from erdl/sensors:master into matt/sensors:37-turn-egauge-notebook-into-script, so I'm noting it here.

See merge commit at https://github.com/matthew-schultz/sensors/commit/951b3d98724b898d2272c2fed464905fd0b7fa21

matthew-schultz, 2018-09-25:
I'm going to go ahead and close this issue since we merged its related PR a while ago.

closed 2018-09-25 by matthew-schultz