erdl/lonoa #39: CCS811 Unable to communicate over I2C

jygh98, 2018-05-19:
So over the past couple days i have been working on trying to get all the air quality sensors (ccs811, bme280, pms5003, and bmp280 for extra) ready for deployment/logging.  However today i noticed while working on the CCS811_logger.py that i am getting nonstop errors and [after looking up the problem up](https://stackoverflow.com/questions/46996135/raspberry-pi-stopped-communicating-with-i2c-sensor) i am not the only stumbling across this problem.  Notice that he is using the same example Adafruit code that i was using.

After going through [the threads on the raspberry pi forum ](https://www.google.com/search?q=ccs811+raspberry+pi+site:www.raspberrypi.org&sa=X&ved=0ahUKEwiKy9zL5JDbAhVJyGMKHaK4DJYQrQIIQigEMAE&biw=1920&bih=987) i soon learned the following the CCS811 sensor relies on i2c clock stretching which the Raspberry Pi does not support very well.  Essentially this is more of a hardware issue in which i am trying to figure a software fix for.  So far i have tried the following as suggested from looking at the forums:

1. I have tried to reduce the baudrate to 10k, 8k, and 4k
2. Going through all the python repositories i can find on github/google

Currently i am hoping for the [owner of the following repository](https://github.com/bitbank2/CCS811) to get back to me on how to compile and run his script.  Along with that i have replied to several threads with similar issues on this sensor and i hoping that someone would be able to get back to me over the weekend.  [Also i may try to play with this sensor of the raspberry pi 3 as this thread suggests that the newer versions support i2c clock stretching.](https://www.element14.com/community/thread/49698/l/i2c-clock-stretching-feature-in-raspberry-pi-3?displayFullThread=true)

As to how only all of a sudden the ccs811 stopped communicating, people suggested that it could be due to this "burn-in period" which i dont completely understand how that could affect it.

Resources:
[What is the i2c clock stretching problem with the raspi 2](http://www.advamation.com/knowhow/raspberrypi/rpi-i2c-bug.html)
[adafruit support telling the customer to just straight up by a different sensor](https://github.com/adafruit/Adafruit_CCS811_python/issues/1)

jygh98, 2018-05-19:
@carlosparadis

jygh98, 2018-05-19:
After letting this issue sink in for a bit, i am suspecting that this issue is likely can be fixed by running the script on 9S, which is a raspberry pi 3.  The reason i think this is because the [i2c clock stretching issue is native to the Broadcom BCM2835 cpu](https://github.com/raspberrypi/linux/issues/254) which is used in the raspberry pi 2.  As linked above, it appears that the raspberry pi 3 supports i2c clock stretching with the newer BCM2837 cpu.

carlosparadis, 2018-05-19:
I'm diminishing the priority of the Issue to normal given it only affects the sensor measuring co2 and vog, neither of which are required in the air quality station project or hobo.

Lets save the label to a pi dying to a mistake or getting compromised. 

Beyond terminology, and given the time left, I believe you should prioritize:

   * Data quality of the sensors required on both projects by comparing to hobo data and local AIR quality sensors
   * Stress tests to see if there are no missing data or time lag between the sensors timestamp 
   * Crontabs, review location to save them so @ryantanaka knows when you are gone, etc.

I think it would be a good idea to also add to all sensor readings, besides the min averages to be saved, also a max and min timestamp at min level for quality checking.

jygh98, 2018-08-21:
I will try rerun the script and test if i am able to log the CCS811 sensor without issue on the raspberry pi.  If not, it seems like this is a hardware issue between the sensor and raspi and can be fixed using an arduino which offers i2c clock stretching.  I do have a spare arduino that i wouldnt mind giving to the lab to use.

jygh98, 2018-08-21:
I and Carlos talked, and we will see if there is enough time left up to the end of next week, we can try and set this running in the Arduino, otherwise this will be left as future work.