erdl/lonoa #40: Conduct Stress Tests on 9S and 2B PI's

carlosparadis, 2018-05-19:
You should leave them running overnight with all the sensors ready for both units to ensure they will handle it. I am particularly worried given our concerns on issue #17. 

Do you think you could do this before you leave? Things you should watch for are for example:

   - Lagged timestamps (are all readings logging at the same time over time, or are they varying (e.g 1s 2s 1s 2s 3s deltas for the low sampling resolution you are using to average could lead to biased measurements).

   - PI temperature (can it fry an egg?) 

Etc.

jygh98, 2018-08-21:
Running temperature as of today
2B : 42.2*C
9S : 51.5*C

Temeprature appears to be running at an [ideal temperature](https://www.reddit.com/r/raspberry_pi/comments/3z50sp/what_is_the_maximum_safe_temperature_noob/) and shouldnt be a concern.  Interesting to note that 2B runs a full 10*C cooler than 9S, even though 2B is the older raspberry pi.

In terms of logging script stability, only issue so far is #48 and i will need to follow up with ryan on that.

carlosparadis, 2018-09-08:
For the record, there was not enough time to finish this one, however the main concern for the stress test was to see if the PI was gonna fry, which has been fine for now over 3 months, so its OK.

There is still a minor missing request here, so I will leave the issue open for now.

closed 2018-09-08 by carlosparadis