erdl/lonoa #41: Conduct Quality Tests on 9S and 2B PI's sensors

carlosparadis, 2018-05-19:
After all sensors are put together in the 2 units, it would probably be healthy to check how it fares:

 * Against the HOBO (in the case of the HOBO project) 
 * The CO2 expensive sensor eileen bought (VOC+CO2 to get it working)
 * Local website reports of air quality and PM 2.5, 1, etc (won't be as precise as the other two sensor comparisons, but at least you can see if it is not completely off the mark to the point it is absurd).

jygh98, 2018-08-21:
This issue currently relies on #39 to be resolved first.

carlosparadis, 2018-09-08:
For the record, we went over this, and the python notebook contains the necessary information to see already some sensors are a little strangely off, and also we don't have the code, but it should not be hard to replicate the experiment that @jygh98 did.

closed 2018-09-08 by carlosparadis