erdl/lonoa #47: augment egauge api script to now use state timestamps

ryantanaka, 2018-06-05:
The script developed in #37 should now use state timestamps:

- script reads the most recent time from a state file
- pulls data for the interval `[ read time, current time)`
- upon successful data acquisition, append `current time` to timestamp file

The script, when run again, should refer to this state file. The idea is to eventually have this script being run by a cron job over and over.

carlosparadis, 2018-06-05:
@ryantanaka I believe for the purpose of the system you will want to stick to unix timestamps, which disambiguate timezones. It should still provide a time window, nonetheless, in case someone wants to use it to gather specific data I guess.

ryantanaka, 2018-06-05:
@carlosparadis, right, I went over using unix timestamps for this with @matthew-schultz but didn't specifically write it up there. Will edit the issue message.

matthew-schultz, 2018-06-23:
Next steps after talking with Carlos:
- thoroughly test for any overlapping egauge values (create an expected output file over a large period of time (eg. January to March) and simulate rerunning the script at smaller increments over that time. Diff the output to confirm that it ran correctly.)
- Avoid creating functions as much as possible so we can trace the program structure more clearly from start to end.

carlosparadis, 2018-07-06:
@matthew-schultz Hey Matt, I see you have sent a lot of commits to your fork, could you give me a few words on how it is coming along?

matthew-schultz, 2018-07-07:
@carlosparadis it's going pretty well. I was able to get the egauge api tests working today with help from  @ryantanaka, so for now I'll probably just be trying to clean up the code to make it easier to understand. 

If you want to look over the egauge api script and test class they're both in the scripts directory here https://github.com/matthew-schultz/sensors/tree/47-egauge-timestamps/egauge/script.

I will probably make a pull request later from that branch in my sensors repo later today

matthew-schultz, 2018-07-26:
I just noticed my process of pushing test branches and force pushing branch updates to github seems to have created numerous ```added a commit...``` messages that don't seem to disappear after a force push.

I might have to rethink my commit process, so the issues don't get so cluttered in the future.

carlosparadis, 2018-07-26:
@matthew-schultz yep, you see why I told you about spaghetti concerns and abandoning PR 50. You mainly want to keep local dev going in a separate branch, and only sync another branch to the PR. 

Assuming you are not doing anything too crazy, the merge will be a "fast forward", which won't prompt a commit message saying merge, and won't lead to conflicts.

matthew-schultz, 2019-02-13:
Closing this issue since https://github.com/erdl/sensors/issues/49 is the latest egauge issue

closed 2019-02-13 by matthew-schultz