erdl/lonoa #48: PMS5003_logger.py on pi 2B error or not?

ryantanaka, 2018-06-12:
Looks like there may be an issue with the sensor.

Current Crontab Setup:
`* * * * * /home/jason/sensors/raspberry_pi/pms5003/pms5003_check.sh`

So it should have been running, however the `pi_pms5003_log.csv` wasn't being updated. 

Running the `PMS5003_logger.py` by itself threw the following exception:

```
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 490, in read
    'device reports readiness to read but returned no data '
serial.serialutil.SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "PMS5003_logger.py", line 99, in <module>
    main()
  File "PMS5003_logger.py", line 64, in main
    s = ser.read(FRAME_LENGTH)
  File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 497, in read
    raise SerialException('read failed: {}'.format(e))
serial.serialutil.SerialException: read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
```
For some reason, this could be because I tried to run one instance of the program manually while the crontab was trying to run (maybe this relates to the msg: `multiple access on port?`. Oddly, the program is now running and updating the csv so I'm not sure what was going on..

jygh98, 2018-08-21:
@ryantanaka I will follow up on you either on Thursday or Friday this week about this.

carlosparadis, 2019-05-21:
Jason graduated, I am transferring this issue to myself.