erdl/lonoa #49: update egauge script to insert latest sensor readings into database

matthew-schultz, 2018-06-23:
@carlosparadis @ryantanaka 

Add functionality to sensors/egauge/script that will access postgresql database using the sqlalchemy Python3 library and update it with the latest sensor readings.

- retrieve egauge sensor data (sensor id, database id, last_run_time) from database
- insert latest data downloaded from egauge api
- truncate all datetime objects to seconds to avoid dealing with fractions of a second
- make the whole process atomic for each egauge sensor (will not update that sensor's data if any errors are thrown)

matthew-schultz, 2018-06-23:
I'll start working on this issue once I finish the preceding steps listed in the Matt Tasks issue https://github.com/erdl/sensors/issues/25

carlosparadis, 2018-06-23:
thanks @matthew-schultz !

carlosparadis, 2018-07-25:
A note for the future here:

Webctrl requires username/password to login. When the time comes, we can create a third and global table in the database, authentication_tokens or w/e name, and store all the tokens we need and to whom it belongs. This should avoid another config file.

matthew-schultz, 2018-07-26:
@carlosparadis @ryantanaka 

So far I've been able to figure out how to connect to a postgresql database using just the database name like so:
``` engine_string = 'postgresql:///DATABASE_NAME' ```

Is it okay to have the database name hardcoded like that in api_egauge.py?

# More details:

These are just my initial impressions working with the SQLAlchemy ORM, so these details might not be 100% correct but:

It seems like its initial connection to the database usually needs:
- type of database system (postgresql)
- orm adaptor (pyscopg2)
- username
- password
- host
- port
- database name

However if we have an environment with postgresql set up like I have it currently, it seems like we just need the database system and name since it appears that:
- username defaults to using the current Linux user, so if they have proper permissions the password isn't needed
- host and port use the default localhost values
- postgresql uses the Psycopg 2 engine by default

It looks like we have https://github.com/erdl/survey_admin/blob/412ceeb738665f9cd699b8ace6fb8a46ad230f86/survey_admin/database.py#L10 as an example for connecting to the database, but that method seems to require more than just the database name

```python3
url = 'postgresql://' + data["user"] + ":" + data["pass"] + "@localhost" + ":" + data["port"] + "/" + data["dbname"]
```

carlosparadis, 2018-07-27:
@matthew-schultz I don't see a problem. I want you to try and see if the database will force you to use a username and password or if it can assume defaults from the server. Let me know.

matthew-schultz, 2018-07-28:
@carlosparadis Accessing the database using just the database name seems to be working well, so far. I plan to test that more thoroughly by creating a minimum permissions Linux user on my machine to run the crontab that runs api_egauge.

What do you think if I implement the prototype egauge script using these tables:
# Proposed table structure
[image]

Tablename: **timestamp_log**
Columns: id(int), ts(timestamp), sensor_id(varchar), status(boolean)
Use: keep a record each run of api egauge for each sensor. Insert 'True' into status after a successful run or insert 'False' if not.

Tablename: **sensor**
Columns: sensor_id(varchar), unit(varchar)
Use: obtain a list of sensors and sensor information. Iterate through list to make an egauge api call and insert that data for each sensor.

Tablename: **reading**
Columns: reading_id(int), datetime(timestamp), sensor_id(int), reading(double), upload_ts(timestamp)
Use: store the values of the sensor readings.

carlosparadis, 2018-08-02:
@matthew-schultz For now, just focus on the basic functionality. purpose_id never served its true purpose, which was giving Eileen ability to change without editing code. 

At the end of the day, because of the sea of configs, she always had to ask someone who was a programmer to edit it for her. 

As we discussed:

 * One table to log timestamps of database insertion, whether it was successful or not (call the table **database_insertion_timestamp**, I dont think it can get any more obvious than this by the name)
    *  timestamp | is_sucess  (feel free to use data types DATETIME and BOOLEAN to keep it simple for now) 
 * One table to log the readings, and the sensor_id. (serial and integer) ; call the table **reading** as you already did. 

Note you do not need to manually create a table that those id refers to in your prototype, because your code never should care about them anyway. This is for eileen purposes only. 

Remember the main focus right now is wrap up the two set of sensors, not all this side functionality that is common to them all. These will warranty some design decisions that are best done after we have the minimal code and logger functionality for them all. 

I am not sure at this point if it is worth an FK between the reading and insertion attempt. Granted, all the rows in readings will refer to success attempt, and this is not the main purpose of the table alone. We also care about what was not inserted. In that sense, most queries will be done by Eileen when she spots **missing data**. This means she will likely be doing queries directly to the table **database_insertion_timestamp**. 

I see you like the white board. One useful thing for you to take notes of is what kind of queries the group can make out of it, instead of focusing on the usual table cardinality. 

But again, try to stick to the bare bones for now.

matthew-schultz, 2018-08-03:
@carlosparadis 

Thank you for that input! It looks like I was able to implement just a little more than we actually needed for the prototype haha. I should be able to make a pull request with a working prototype with those modifications later today (8/3) or monday (8/6).

I'll try to keep in mind what queries a user could make when designing tables, and I'll go with your two table __database_insertion_timestamp__ and __reading__  structure for the prototype.

__For the prototype do we want to parse out just one value from the different values in one row of sensor/datetime readings for insertion into reading?__

E.g.
for an example egauge api row retrieved at time 1533329480 for sensor 725:

Date & Time | Usage [kW] | Phase A [kW] | Phase B [kW] | Phase C [kW]
--- | --- | --- | --- | ---
1533001320 | 53.11 | 16.44 | 18.58 | 18.076016667

would be inserted as this row in __reading__:

id | sensor_id |datetime | unit | value | upload_timestamp
--- | --- | --- | --- | --- | ---
1 | 725  | 1533001320 | Usage [kW] | 53.11 | 1533329480

---
# Proposed prototype functionality:
- make an egauge api request for one sensor using the latest timestamp received from the database and the current time
- parse out one row of reading data for each row returned by the api request
- insert each row's parsed data into __reading__
- insert a row into __database_insertion_timestamp__ marking success or failure of insertion of data into reading

matthew-schultz, 2018-08-07:
@carlosparadis 
My branch https://github.com/matthew-schultz/sensors/tree/49-sql-alchemy should now have a complete working egauge prototype with database functionality.

Should I go ahead and make a pull request?

carlosparadis, 2018-08-07:
@matthew-schultz I want you this time around to try and make the PR, and **before submitting it** take a screenshot of the window that shows the diff, and attach here. 

Lets do this exercise this time, so we dont need to ammend the gigantic PR :) I just want to make sure we are on the same page on what I mean by this too.

carlosparadis, 2018-08-07:
@matthew-schultz sorry, just realized I missed your 3 day comment! I will go over it still tonight, but could you tell me what you decided to go with on the questions you made to me?

matthew-schultz, 2018-08-07:
@carlosparadis no worries!

I went ahead and implemented the __reading__ table pretty much how I explained in the question. So for each row of sensor readings, I insert one row into the table  kind of like this:

reading_id | sensor_id |datetime | units | reading | upload_timestamp
--- | --- | --- | --- | --- | ---
serial | int | timestamp | varchar | double | timestamp
1 | 725  | 1533001320 | Usage [kW] | 53.11 | 1533329480

I expect that the full version may parse out multiple readings from a single row returned by the api, but I think this is a good start for the prototype.

I should be able to get that pull request diff screenshot too; might have to take multiple screenshots though.

matthew-schultz, 2018-08-07:
@carlosparadis So the diffs are pretty much complete files since I renamed both the old files and added one new one. (>_< ''')

Should I rebase and separate out the commits renaming a file from the commits updating a file, so it's easier to see what changed?

matthew-schultz, 2018-08-07:
Here's a screenshot of a branch where I didn't rename the files, if that makes it easier to see what changed vs. what stayed the same. Sorry it's probably a lot to process at once~

[image]

carlosparadis, 2018-08-07:
@matthew-schultz Send me the screenshot :^)

carlosparadis, 2018-08-07:
@matthew-schultz 

> For the prototype do we want to parse out just one value from the different values in one row of sensor/datetime readings for insertion into reading?

I don't understand this question. What goes on reading is what the notebook parses for now. Was that what you did?

The commit labels need to be more descriptive. Remember I told you to focus on what .git doesn't tell you. I would know what and where you changed the commits even without your current label. You need to relabel them to say why.

Give it a shot on the labels here on text as a comment!

matthew-schultz, 2018-08-08:
@carlosparadis It might be easier for me to explain over a call some time?

One row of readings in the notebook looks like this

Date & Time | Usage [kW] | Generation [kW] | Other [kW] | Water heater [kW] | Whole house [kW] | ACC [kW] | AHU [kW] | Dryer [kW] | Range [kW] | Kitchen refrigerator [kW]
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
1505352240 | -0.0 | -0.0 | 0.477750 | 0.069967 | 1.953250 | 1.252933 | 0.138517 | 0.012717 | 0.000867 | 0.000500

The non-'Date & Time' column names can differ from sensor to sensor.

If you want it to look exactly the same when it is inserted into the __reading__ table, it looks like I will need to hardcode columns in __reading__ for each specific sensor (e.g. Usage [kW], Generation [kW], Other [kW], etc).

I would prefer not to hardcode values like that since they can differ from sensor to sensor.

__I could change the program parse out each value like this__, so the prototype would be more flexible and work for more sensors as long as the Reading is a double:

datetime | reading | units
-- | -- | --
1505352240 | -0.0 | Usage [kW]
1505352240 | -0.0 | Generation [kW]
1505352240 | 0.477750 | Other [kW]
1505352240 | 0.069967 | Water heater [kW]
1505352240 | 1.953250 | Whole house [kW]
1505352240 | 1.252933 | ACC [kW]
1505352240 | 0.138517 | AHU [kW]
1505352240 | 0.012717 | Dryer [kW]
1505352240 | 0.000867 | Range [kW]
1505352240 | 0.000500 | Kitchen refrigerator [kW]

Do you want something like that?

__

# Current version in https://github.com/matthew-schultz/sensors/tree/49-sql-alchemy

__Only parses the first value in each row__

The readings from the api request:

Date & Time | Usage [kWh] | Generation [kWh] | Other [kWh] | Water heater [kWh] | Whole house [kWh] | ACC [kWh] | AHU [kWh] | Dryer [kWh] | Range [kWh] | Kitchen refridgerator [kWh]
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
1505372400 | 0.0 | 0.0 | 914.677966 | 40.667430 | 3374.120953 | 1631.121273 | 179.117595 | 326.280238 | 207.852151 | 74.404300
1505368800 | 0.0 | 0.0 | 914.086887 | 40.664147 | 3372.287279 | 1630.010558 | 179.029650 | 326.277871 | 207.851150 | 74.367017
1505365200 | 0.0 | 0.0 | 913.546429 | 40.660866 | 3370.445513 | 1628.879968 | 178.892024 | 326.276184 | 207.850229 | 74.339813
1505361600 | 0.0 | 0.0 | 913.019358 | 40.656918 | 3368.592838 | 1627.717554 | 178.754530 | 326.274547 | 207.849284 | 74.320645
1505358000 | 0.0 | 0.0 | 912.575539 | 40.649671 | 3366.769215 | 1626.518802 | 178.616350 | 326.272947 | 207.848344 | 74.287562
1505354400 | 0.0 | 0.0 | 912.150748 | 40.628782 | 3364.565113 | 1625.286709 | 178.477743 | 325.934073 | 207.847400 | 74.239658

Are inserted like this:

reading_id | sensor_id | datetime | reading | units | upload_timestamp
-- | -- | -- | -- | -- | --
1 | 725 |1505352240 | -0.0 | Usage [kW] | 1533329480
2 | 725 | 1505352180 | -0.0 | Usage [kW] | 1533329480
3 | 725 | 1505352120 | -0.0 | Usage [kW] | 1533329480
4 | 725 | 1505352060 | -0.0 | Usage [kW] | 1533329480
5 | 725 | 1505352000 | -0.0 | Usage [kW] | 1533329480
6 | 725 | 1505351940 | -0.0 | Usage [kW] | 1533329480

matthew-schultz, 2018-08-08:
How do these commit messages sound?

- Build models for database tables and functions to create and drop tables for testing
- Update script for database retrieval and insertion instead of text files
- Update unittest to work with database instead of text files
- Rename api script to fit naming scheme PURPOSE_SENSORTYPE.py
- Rename unittest script to fit naming scheme PURPOSE_SENSORTYPE.py

carlosparadis, 2018-08-10:
* define ORM to store database insertion success and readings
 * replace api requests storage from text to database
 * replace unittest from text to database

Is this what you are doing on the first 3? 

I am not sure what are the last two bullets.

Also, I suggest you branch out the webctrl code from the current accepted branch in erdl, and work in webctrl on the meantime so we can still hit the Aug 15 deadline for both.

matthew-schultz, 2018-08-10:
@carlosparadis 

> Is this what you are doing on the first 3?

Yeah that sounds about right; I'll use those commit labels if that is okay with you

>> - Rename api script to fit naming scheme PURPOSE_SENSORTYPE.py
>> - Rename unittest script to fit naming scheme PURPOSE_SENSORTYPE.py

> I am not sure what are the last two bullets.

We discussed renaming a couple files over slack for consistency in the naming scheme; maybe this makes more sense:
- Renamed egauge_api.py to api_egauge.py for consistency
- Renamed test_egauge_api to test_egauge.py for consistency

I just want to confirm. Are you okay with how I parse and insert data into the __reading__ table in my  [comment](https://github.com/erdl/sensors/issues/49#issuecomment-411534683) from 8/10?

carlosparadis, 2018-08-10:
@matthew-schultz Oh I see. yes, the commit names are good now :) I know what you are talking about.

The readings should be in long format, as you did. This is how in the database it is stored. The python notebook was on wide-format to facilitate life on Tableau, I had forgotten that.

Could you post an example of the final tables we have being implemented? I know we agreed on two but I am getting confused if there is more from the last posts.

matthew-schultz, 2018-08-10:
@carlosparadis You had me scared for a moment that the table really had to be wide format haha

Here are what the two tables currently look like:

---

__reading__

reading_id | sensor_id |datetime | units | reading | upload_timestamp
--- | --- | --- | --- | --- | ---
serial | int | timestamp | varchar | double | timestamp
1 | 725  | 1533001320 | Usage [kW] | 53.11 | 1533329480
---
__database_insertion_timestamp__

id | timestamp | is_success
--- | --- | ---
serial | double | boolean
1 | 1533329420 | True

carlosparadis, 2018-08-10:
@matthew-schultz Just call both of the columns **timestamp**. Feel free to do the PR afterward!

matthew-schultz, 2018-08-11:
Some notes about objects that store time in Python and Postgresql as they relate to this project:

In __Python__, I prefer using the __arrow__ data type for times because they are easily manipulated and converted to different types of time data types (e.g. datetime, time, integer/string/epoch).

The current __Postgresql__ tables use the __TIMESTAMP__ data type for any times.

The Python __ORM__ sqlalchemy I'm using to connect the python script with the database uses __datetime__ Python objects when inserting or reading 'TIMESTAMP' Postgres data.

The __times returned by making api requests to the egauge host__ are __strings__ that can be read as __integers__ representing the number of seconds elapsed since the Linux epoch __or arrow.timestamp()__ objects

---

I'm going to try to use this naming scheme

- arrow objects: '_time'
- datetime objects: '_datetime'
- seconds elapsed since Linux epoch: '_timestamp'
- postgresql TIMESTAMPs: 'timestamp' or '_timestamp'

matthew-schultz, 2019-01-26:
## Outline of proposed api_egauge update
In order to insert readings obtained from one sensor_id that may have multiple purposes, I'm planning on structuring api_egauge to:
- get a list of __sensor_ids__ and iterate through those in __main__
- get a __sensors__ list in __get_data()__ by querying all sensor_info rows where sensor_id matches
  - include purpose_id, last_updated_datetime, and sensor_part in each sensor item
- iterate through the __sensors__ list in __insert_readings()__

```python3
if _name_=="main":
   conn = get_db_handler()
   # get a list of unique, active egauge sensor_ids
   for each sensor_id in sensor_ids
      readings, sensors = get_data(conn, sensor_id)
      insert_readings(conn, readings, sensors, sensor_id)

def get_data(conn, sensor_id):
   #get a list of sensor_info objects (sensors) that match sensor_id, including purpose_id, last_updated_datetime, sensor_part columns
   #find earliest last_updated_datetime in sensors
   #make egauge api request using earliest last_updated_datetime
   for sensor in sensors:
       #insert status into error_log table
   return readings, sensors

def insert_readings(conn, readings, sensors, sensor_id):
   for sensor in sensors
      # insert all readings after sensor.last_updated_datetime if sensor_part matches
      # insert status into error_log table

matthew-schultz, 2019-02-13:
Notes about next update to Pull Request https://github.com/erdl/sensors/pull/58 :
- Update script to use latest schema defined here: https://github.com/erdl/sensors/issues/62#issuecomment-452172935
- Insertions into readings table should be atomic on the sensor_id level 
   - all purpose_ids associated with a given sensor_id should have the same sensor_info.last_updated_datetime
- For consistency, use pendulum python library for time instead of arrow

carlosparadis, 2019-05-21:
Closing as I accepted the egauge PR #58.

closed 2019-05-21 by carlosparadis