erdl/lonoa #51: Create script that inserts webctrl readings into database

matthew-schultz, 2018-08-10:


matthew-schultz, 2018-08-10:
@carlosparadis I just created this issue; will be working on implementing this for now

: D

matthew-schultz, 2018-08-14:
@carlosparadis 

From my understanding of the webctrl api, each request requires a date and it returns all readings for that date (at 5 minute intervals from 12:00 am to 11:55 pm inclusive during that date).

Is it okay if I build the script with these assumptions? : 
- webctrl sensors store readings for every 5 minutes
- the api_webctrl script will store timestamps at the __date__ granularity in the __database_insertion_timestamp__ table
- api_webctrl will make an api request only if the timestamp in the __latest success row in database_insertion_timestamp__ < __current date__
- the readings will only be inserted __if 288 rows have been returned by the api request__ (one for every five minute interval recorded during a given date) and if the __last reading's timestamp__ == __11:55 PM on the latest date__ in __database_insertion_timestamp__
- a success row with a timestamp of (the latest read success row + 1 day) will be inserted into database_insertion_timestamp __only if the above conditions are met__

carlosparadis, 2018-08-14:
Didn't egauge also required a start and end timestamp? Just want to make sure I am not misunderstanding your first question.

Tell me what you see different in respect to what already discussed in egauge, I thought it would be the same, aside from maybe the sampling resolution.

matthew-schultz, 2018-08-14:
@carlosparadis 

The webctrl api seems to only take dates like '2018-08-13' for the start and end times; egauge took timestamps all the way down to the second like 2018-01-01 12:01:00, 

So since webctrl only takes dates and running the script successfully multiple times in one day would reuse the same date argument, I've been thinking that I will probably either have to:

1. run the script every five minutes and filter out readings that have already been inserted for the current date
2. run the script once a day and insert all the readings for one day all at once

Should I go with option 1?

carlosparadis, 2018-08-14:
@ryantanaka has a cron job for webctrl running at minute level. @ryantanaka, can you tell us what you did? 

@matthew-schultz For now, carry on with option 1 and re-evaluate afterward ryan respond if there is a better way. You can always try and find on legacy-scrape-util how it goes about that, but I don't think it is a good idea to spend time reading the code if you can't find in less than 5 or 10 minutes at most.

matthew-schultz, 2018-08-14:
@carlosparadis 

I have an initial version of the script running on my local machine using 3 tables:

---

__webctrl_user__

id | username | password
-- | -- | --
---
__reading__

reading_id | sensor_path | timestamp | units | reading | upload_timestamp
--- | --- | --- | --- | --- | ---
serial | varchar | timestamp | varchar | varchar | timestamp
205 | ABSPATH:1:#frog1_room_sensors_1/zone_temp_1_tn | 2018-08-13 07:05:00 | a     | 0.9     | 2018-08-13 17:26:00
 
---
__database_insertion_timestamp__

id | timestamp | is_success
--- | --- | ---
serial | timestamp | boolean
7 | 2018-08-13 17:35:00 | t

---

How do those look so far?

matthew-schultz, 2018-08-15:
Script differences relating to database_insertion_timestamp
---
The __egauge__ script currently uses  __database_insertion_timestamp__ to store the next timestamp it will use as the start value in an api request (corresponding to a row in __reading__ that __has not yet been inserted__). I implemented this design mainly because during my testing, egauge api requests at the minute granularity retrieved every reading from the start time up to but not including the end time.

The __webctrl__ script currently uses __database_insertion_timestamp__ to store the timestamp for the last row that __has been inserted__ into __reading__. The next api request will only insert rows into __reading__ with later timestamps. I implemented this design because during my testing, webctrl api requests retrieved every reading from the 12:00 am of the start date up to and including the last elapsed 5 minute increment of time on the end date.

matthew-schultz, 2018-08-15:
@carlosparadis 

I'm currently testing working versions of orm_webctrl.py and api_webctrl.py, but before I make a PR for those  how do these commit messages sound?

- Define ORM to store database insertion success and readings
- Create script to obtain readings from webctrl sensors and insert into db

matthew-schultz, 2018-09-22:
@carlosparadis Before I make the PR do these commit messages sound okay?

- Define ORM to store database insertion success and readings
- Create script to obtain readings from webctrl sensor and insert into db

carlosparadis, 2018-09-22:
`Define ORM to store database insertion success and readings`

"Store database insertion" doesn't make a lot of sense for me. Just call it the ORM for Webctrl data manipulation. I suppose the commit also convey insertion failures, correct?

matthew-schultz, 2018-09-22:
That's correct, that commit also accounts for the database storing insertion failures.

Do these commit messages sound good?
```Define ORM for Webctrl data manipulation```
```Create script to obtain readings from webctrl sensor and insert into db```

matthew-schultz, 2018-09-22:
Ah thank you for clarifying that information about webctrl being an interface not a sensor~

How about these commit messages?
- Define ORM for Webctrl data manipulation
- Script to insert webctrl readings into db

From my count, they should fit the <50 character limit

carlosparadis, 2018-09-22:
I think you still want the "get" stuck somewhere in there :v since that is an expected piece of the pipeline. Maybe:

`Script to get and insert webctrl readings into db` ?

matthew-schultz, 2018-09-22:
Sounds good and the word count checks out! : D

I'll go ahead and use that message

carlosparadis, 2018-09-22:
Thanks! Sorry for being overzealous on the wording :) I will worry way less once we are past the basic functionality, promise!

matthew-schultz, 2019-02-13:
Notes about next update to Pull Request https://github.com/erdl/sensors/pull/59 :
- Update script to use latest schema defined here: https://github.com/erdl/sensors/issues/62#issuecomment-452172935
- For consistency, use pendulum python library for time instead of arrow

carlosparadis, 2019-05-21:
Closing as PR #59 was accepted today. We should spread remaining requirements over issues later.

closed 2019-05-21 by carlosparadis