erdl/lonoa #60: Create script that inserts HOBO readings into database

matthew-schultz, 2018-09-25:


matthew-schultz, 2018-09-25:
@carlosparadis 

What do you think about designing the main block for the hobo script like this:
```python3
if __name__=='__main__':
    hobo_path = '9790163-sample.csv'
    conn = get_db_handler()
    try:
        latest_timestamp_successfully_inserted, readings = get_data_from_folder(conn, hobo_path)
    except Exception as e:
        log_failure_to_get_data()
        sys.exit()
    try:
        insert_readings_into_database(conn, latest_timestamp_successfully_inserted, readings)
    except Exception as e:
        log_failure_to_insert_readings()
        sys.exit()
```

carlosparadis, 2018-09-25:
@matthew-schultz Looks good to me!

carlosparadis, 2018-09-27:
Frankly, we should pick better names for all 3. api_webctrl is not suggestive we insert in the database either. Maybe we should eventually call all them by just their names: egauge.py, webctrl.py, hobo.py. 

But I feel it's too early to worry about these details, so feel free to go ahead and choose either extract or read.

matthew-schultz, 2018-09-28:
@carlosparadis 
I've got an initial version of the HOBO script working, so I might be able to do a pull request soon

But I made some modifications to the main block, so I just wanna get it checked one more time:
- get_data_from_folder() is now get_data_from_hobo_csv() so it's a little more precisely named
- returned more variables from get_data...() like hobo_id, latest_timestamp_successfully_inserted, readings, and units since I use them in insert_readings_into_database()

```python3
if __name__=='__main__':
    hobo_csv_filename = '9790163-sample.csv'
    conn = get_db_handler()
    try:
        hobo_id, latest_timestamp_successfully_inserted, readings, units = get_data_from_hobo_csv(conn, hobo_csv_filename)
    except Exception as e:
        log_failure_to_get_data()
        sys.exit()
    try:
        insert_readings_into_database(conn, hobo_id, latest_timestamp_successfully_inserted, readings, units)
    except Exception as e:
        log_failure_to_insert_readings()
        sys.exit()
```

carlosparadis, 2018-09-28:
@matthew-schultz Actually your function should be scanning a specific folder for any csv put into it to then insert into the database, so the previous name was as specific as it could get. 

Thats how Eileen is used to do that with scrape-util. 

There is one more thing to consider on what to do with the files if insertion fails or succeeds since these are obtained from the server instead of an api that keeps the data for us. You should assume the data from the HOBO will be wiped out as well to allow for new data to be sampled on it, so the server is the last place one can obtain it. 

Do some thought on this and let me know, but again I think you want to stick to the `get_data_from_folder()` or more precisely `get_csvs_from_folder()` would likely be ideal.

matthew-schultz, 2018-09-28:
Hmmm, if the final script will be scanning a directory for any csv's what about generally restructuring it like this (without the variables and exception handling just to keep it more simple):
```python3
conn = get_db_handler()
filenames = get_csv_filenames_from_directory()
for filename in filenames:
  try:
    get_data_from_csv()
  try:
    insert_readings_into_database()
```

carlosparadis, 2018-09-28:
@matthew-schultz 

I prefer the function to still remain `get_csv_from_folder()`. Eileen and Kat know that by putting data in a "hobo folder", the data will be eventually inserted into the database. So having it explicit the word folder in the main code will make it familiar for a future programmer talking to them diagnose where the problem is. 

Now, this implies another question we should get on. We can either go the route of:

`get_csv_from_folder()` 

OR

`get_all_csv_from_folder()`. 

I.e. we can either take a "try to insert as many into the database" policy, or a "all or nothing" policy. Thoughts on pros/cons?

matthew-schultz, 2018-09-28:
Ahh that makes sense; I'll use 'folder' instead of 'directory' since that should make it clearer for everyone.

Also, I prefer attempting to have each file's insertion atomic. So even if one file fails, the other ones could still successfully be inserted. That's one of the main reasons why I put try...get_data() and try...insert_readings() statements in a for loop that iterates through a list of files.

```python3
conn = get_db_handler()
csv_filenames = get_csv_filenames_from_folder()
for csv_filename in csv_filenames:
    try:
        get_csv_data_from_folder()
    try:
        insert_readings_into_database()
```

But, I think it that partially depends on how we keep track of successful insertion of an individual csv's data into the database.

Do you know if there's a specific way we do that right now or that we want to implement?

carlosparadis, 2018-09-28:
> But, I think it that partially depends on how we keep track of successful insertion of an individual csv's data into the database. 

Yes, I see you prefer each file policy but I want you to weight the impact on this from the perspective of someone debugging. 

Right now its pure chaos. Try to describe the pros/cons on both approaches from a debugging perspective as an exercise, I wouldn't use the previous one as the reference.

Also while at it, weight in what happens to files that get inserted successfully, those that are not due to errors, and those that are still pending any attempts.

carlosparadis, 2018-10-02:
In respect to your last comment, though luck. You would have a hard time troubleshooting which 1234.csv of what day was the actual one (try imagining Eileen going over this by memory). 

In scrape-util we went on with 3 folders, but I feel there is still room for improvement. I will catch up on your prior comment most likely tomorrow or W!

matthew-schultz, 2018-10-02:
Ahh for some reason I was assuming that each csv filename would be unique.

If I can't assume each filename will be unique, what if I add a function that renames a csv based on the timestamp of its last reading row? (place _YYYYMMDD-hhmm between filename and file extension)

E.g. __1234.csv__ becomes __1234_20181001-2300.csv__ if its last row of readings has the timestamp 2018-10-01 23:00.

carlosparadis, 2018-10-09:
Sorry, I am late on this. I like the renaming option, but the problem is that this forces some words that can never be used on file names that could affect the behavior of the script. Sounds more trouble on the long-run to remember. 

I think we should move to think out of the box on this matter by adding some redundancy to what we call the id for every file. 

 * For instance, one thing that is immutable to the file, and I would dare say **unique** in the context of our application, is the **Creation Date of the file**. This is a file metadata, and someone can trace down the file by looking into this in their own computer folder (careful to not use the **Modified date that changes**). 
 * The second thing is the file name. If someone renames it, then they need to do the more troublesome searching for **Creation Date**. But if they don't, then its easy to find. 
    * Here I think we should just keep one folder rather than 2 or 3. The reason being is in the same folder, then there is no way anyone uploading can make the file with duplicate name: When they try to, it will ask if they want to overwrite! If we however put with another folder, then the script will have to deal with that, which is more troublesome. Because we have no guarantee someone wouldn't go after the second or third folder and edit there (which would mess all the assumptions), one folder seems overall the more sane way. 
 * Last but not least you can and should also add the first timestamp and last timestamp of the **reading inside the file**. This is because when someone sees something messed up in the **readings** table, this is their "gateway" to troubleshoot.

This result in a total therefore of 4 proposed columns:

 * Earliest timestamp of the file (may not be on top, who knows, use max function) 
 * Latest timestamp of the file (may not be on the bottom, who knows, use min function) 
 * The creation date of the file that contained the readings
 * The name of the file

The file name, being in one folder, remains whatever the person wants. When re-reading files, you would assess if the file creation date is or not the same to decide to upload or not. I think this is the most sane approach to avoid the code re-loading the same files sitting in the same folder. If the person tries to duplicate the file and add it in, then it's on the human error side, not our concern atm. 

Why don't you think of some scenarios of troubleshooting and eileen usage to see if these columns suffice for her to quickly diagnose problems and us as well?

matthew-schultz, 2018-10-11:
@carlosparadis 

Yeah, I like the sound of going by the file metadata, filename, and earliest + latest timestamps.

But after looking up how to access the file created date, I might only be able to access the __file modified date__ instead of the __file created date__ especially if we're using python3 on Linux.

> os.path.getctime(path)
Return the system’s ctime which, on some systems (like Unix) is the time of the last metadata change, and, on others (like Windows), is the creation time for path. 

https://docs.python.org/3/library/os.path.html#os.path.getctime

https://stackoverflow.com/questions/237079/how-to-get-file-creation-modification-date-times-in-python

I'll try to go over some troubleshooting scenarios and the HOBO usage with Eileen by the end of the week.

carlosparadis, 2018-10-11:
Ah yes, I forgot this create date issue. Agree, give some more thought on what metadata can serve this purpose, I think you and I are on the same page now on using metadata of the file and it's content to facilitate troubleshooting from Eileen and others from a standpoint it is familiar to them.

carlosparadis, 2018-10-25:
> if either timestamp already exists, insert nothing into reading and insert failure row into hobo_reading_insertion then skip to reading next HOBO csv 

This may be a good opportunity to consider a 3rd status on the log file instead of just success or failure: That of no data to be inserted has been found. I think we didn't consider this status in the previous 2 scripts, but it may be worthwhile. Does it eliminate another annoying question with scrape-util associated to "Is the script running or is it lacking data on the source?"

> I can't really decide what to do with the csv files after that point though. Do we just want to delete them after a successful run?

We leave the file in the folder. This will avoid anyone by accident to upload a file with the same name, and stays there ad-eternum all in the same place. Because the source gets wiped clean, contrary to egauge and webctrl, this one makes sense to, from start, live in the server.

With everything in one folder, it should be doable for Eileen to navigate. The database log will be the go-to as a first check to see what happened instead of the folders anyway. 

Can you please post here the main code for the HOBOs with the function names based on what we agreed? Afterward, we should be good to go.

matthew-schultz, 2018-10-30:
```python3
if __name__=='__main__':
    csv_filenames = get_csv_filenames_from_folder()

    for csv_filename in csv_filenames:
        file_timestamp = pendulum.now('Pacific/Honolulu').format('YYYYMMDD-HHmm')
        conn = get_db_handler()
        try:
            hobo_id, latest_timestamp_successfully_inserted, earliest_file_timestamp, latest_file_timestamp, readings, units = get_csvs_from_folder(conn, csv_filename)
        except Exception as e:
            log_failure_to_get_data()
            continue
        try:
            insert_readings_into_database(conn, earliest_file_timestamp, latest_file_timestamp, csv_filename, hobo_id, latest_timestamp_successfully_inserted, readings, units)
        except Exception as e:
            log_failure_to_insert_readings()
            continue

```

carlosparadis, 2018-11-01:
@matthew-schultz Could you edit the main file alone to showcase a variable with readings, and another called file_metadata being a tuple of the rest? As it stands it may be confusing what all these other information is. Inside the function we can keep them as-is.

matthew-schultz, 2018-11-02:
@carlosparadis So, something like this?

```python3
if __name__=='__main__':
    csv_filenames = get_csv_filenames_from_folder()

    for csv_filename in csv_filenames:
        conn = get_db_handler()
        try:
            file_metadata, readings = get_csvs_from_folder(conn, csv_filename)
        except Exception as e:
            log_failure_to_get_data()
            continue
        try:
            insert_readings_into_database(conn, file_metadata, readings)
        except Exception as e:
            log_failure_to_insert_readings()
            continue
```

carlosparadis, 2018-11-02:
Yes, but could you also show me the return function for get_csvs_from_folder? 

Also, instead of:

> file_metadata, readings = get_csvs_from_folder(conn, csv_filename)

Do:

> file_readings, file_metadata =  get_csvs_from_folder(conn, csv_filename)

That way it is clear the metadata are from the same file that contains the readings from that 1 line of code. :)

matthew-schultz, 2018-11-03:
> Yes, but could you also show me the return function for get_csvs_from_folder?

Something like this?
```python3
def get_csvs_from_folder(conn, csv_filename):
...
    return readings, (earliest_file_timestamp, latest_file_timestamp, file_modified_timestamp, hobo_id, units)
```
I'm glad you mentioned returning a tuple; I almost tried to use a dictionary for that haha

Also wanted to mention that the variables aren't completely finalized since I'm still working on that function (mainly checking if a csv file has been already run successfully or if its earliest and latest file_timestamps are inserted)

---

> Do: 
```file_readings, file_metadata = get_csvs_from_folder(conn, csv_filename)```

Sounds alright, I can go with that order and naming if it makes more sense.

carlosparadis, 2018-11-03:
Yes, this looks great Matt! I wanted to make sure the tuple explicitly showed what is in the file_metadata than some variable that would require reading the entire function code. This will keep it simple for anyone wondering since the return would be the next thing they would check. Make also sure to keep that function well documented since that is the second place someone would go after understanding what and why is the file_metadata returned.

Let me know when you get done with the function then :)

carlosparadis, 2018-11-17:
@matthew-schultz correct

matthew-schultz, 2018-11-19:
@carlosparadis 

I have the pull request just about ready, but I made some modifications to main, so I wanted to make sure they looked okay.

What do you think about adding a __AlreadyInsertedException__ along with a __log_already_inserted() function__ to handle the NO_NEW_DATA_STATUS?

```python3
if __name__=='__main__':
    #get all files ending in .csv in folder
    csv_filenames = glob.glob('./to-insert/*.csv')
    for csv_filename in csv_filenames:
        conn = get_db_handler()
        try:
            file_readings, file_metadata = get_csvs_from_folder(conn, csv_filename)
        except AlreadyInsertedException as e: #inserts status=NO_NEW_DATA_STATUS
            log_already_inserted(csv_filename, e.file_metadata)
            continue
        except Exception as e: #inserts status=FAILURE_STATUS
            log_failure_to_get_data(csv_filename)
            continue
        try:
            insert_readings_into_database(conn, file_metadata, file_readings, csv_filename)
        except Exception as e: #inserts status=FAILURE_STATUS
            log_failure_to_insert_readings(csv_filename, file_metadata)
            continue
```

carlosparadis, 2018-11-19:
I think you still want one function for each condition, then a list of excerpts on main (imagine if we want to add more excepts, the main starts getting obscure). 

What does it mean to fail to get the data? There is no API to get the data here, so I am confused on the intent. If we do want to have more specific excepts to be handled, it would keep things simpler if you had one log function for the major behaviors (process file on the folder, and insert in the database), and then handle the specific cases within each of the 2 functions. In Java one could go about function overload, but I don't remember Python having that. Can always go with the list of excepts in a single function too I guess.

matthew-schultz, 2018-11-19:
@carlosparadis 
> I think you still want one function for each condition

What do you think about adding a third function __check_if_csv_readings_already_inserted()__ and try-catch statement?

I think it would be useful to have one try-catch every section where we want to end the for loop iteration, especially since __check_if_csv_readings_already_inserted()__ would be called fairly often if we never move csv's that have already been read.

```python3
if __name__=='__main__':
    #get all files ending in .csv in folder
    csv_filenames = glob.glob('./to-insert/*.csv')
    for csv_filename in csv_filenames:
        conn = get_db_handler()
        try:
            file_readings, file_metadata = get_csvs_from_folder(conn, csv_filename)
        except Exception as e: #inserts status=FAILURE_STATUS
            log_failure_to_get_data(csv_filename)
            continue
        try:
            check_if_readings_already_inserted(file_metadata)
        except AlreadyInsertedException as e: #inserts status=NO_NEW_DATA_STATUS
            log_already_inserted(csv_filename, file_metadata)
            continue
        try:
            insert_readings_into_database(conn, file_metadata, file_readings, csv_filename)
        except Exception as e: #inserts status=FAILURE_STATUS
            log_failure_to_insert_readings(csv_filename, file_metadata)
            continue
```

> What does it mean to fail to get the data? There is no API to get the data here, so I am confused on the intent.

Maybe a better name would be __log_failure_to_read_data_from_csv()__? The purpose of this except block is to catch and log any exception thrown in __get_csvs_from_folder()__; that way we have log data for debugging.

carlosparadis, 2018-11-20:
It doesn't address the concern that the main function will still be the place to go to add every single exception. I am not worried about performance here, it's a small dataset. The point is to have the logic of error handling based off logical steps on the code behavior, instead of the other way around.

Fundamentally you check files in a folder, and insert into the database. Let the 2 functions to handle errors to these operations do the rest, or the focus of the main function will switch from this, and, remember, only us will be familiar with the "what ifs".

carlosparadis, 2018-11-20:
>             insert_readings_into_database(conn, file_metadata, file_readings, csv_filename)
 

Put file_readings before file_metadata here too.

matthew-schultz, 2018-11-23:
@carlosparadis 

Okay, I should be able to keep main to 2 functions, since we don't want to overfocus main on error handling. However, I would still like to have a continue statement for the 'NO NEW READINGS' status.

Right now I'm thinking about returning a boolean like __no_new_readings__ when I call get_csvs_from_folder(). Then we can either check that boolean within main or within the insert_readings_into_database() function.

What do you think about checking within main like this?

```python3
if __name__=='__main__':
    #get all files ending in .csv in folder
    csv_filenames = glob.glob('./to-insert/*.csv')
    for csv_filename in csv_filenames:
        conn = get_db_handler()
        try:
            #may insert status=NO_NEW_READINGS
            file_readings, file_metadata, no_new_readings = get_csvs_from_folder(conn, csv_filename)
            # skip to next iteration of for loop if file_readings already inserted
            if no_new_readings: continue
        except Exception as e: #inserts status=FAILURE_STATUS
            log_failure_to_get_data(csv_filename)
            continue
        try:
            insert_readings_into_database(conn, file_readings, file_metadata, csv_filename)
        except Exception as e: #inserts status=FAILURE_STATUS
            log_failure_to_insert_readings(csv_filename, file_metadata)
            continue
```

carlosparadis, 2018-11-24:
@matthew-schultz This logic is still a little off. And I think this is a consequence of the function name `get_csvs_from_folder` not reflecting it's true nature.

You see, you do more than just get the csvs from the folder. You also check in the database if the files were already inserted previously or not (otherwise, that `conn` parameter passed over there seems weirdly off. Why would you need a connection to a database to just get csvs from a folder?). 

The function is really something like `get_csvs_not_in_db_from_folder`. 

That being the case, instead of returning a parameter `no_new_readings`, you should have variables `file_readings`, `file_metadata` having value `None` in these cases, and your if statement can rely on that for the logic.

carlosparadis, 2018-11-24:
I am not very fond for continues in replacement of if and else :( 

How about this version instead:

```python
if __name__=='__main__':
    conn = get_db_handler()
    csv_filenames = glob.glob('./to-insert/*.csv')
    for csv_filename in csv_filenames:
        try:
            csv_readings, csv_metadata = get_csv_from_folder_not_in_db(conn, csv_filename)
            try:
                insert_csv_readings_into_db(conn, csv_readings, csv_metadata, csv_filename)
            except Exception as e: 
                log_failure_to_insert_csv_readings_into_db(csv_filename, csv_metadata)
        except Exception as e: 
            log_failure_to_get_csv_readings_from_folder(csv_filename)
```

You can use the `insert_csv_readings_into_db` logic to handle the empty case by checking if `csv_readings` is an empty file. No extra parameters needed, and inside of it have an if and else to decide on what to log into the database, i.e. either sucessful insertion of readings or no readings to be inserted. 

At the end of the day, you want to use that `except` for db insertion both for when the sucessful log into the db fails. This should keep the main very clean and easy to read. 

I would also move the comments of possible errors from the main to the function description of the logs instead. Avoids clutter.

matthew-schultz, 2018-11-24:
What about just one continue, so we can completely skip the second try-catch when get_csvs...() throws exceptions?

I would prefer to avoid calling functions that depend on variables we haven't initialized.

```python3
if __name__=='__main__':
    conn = get_db_handler()
    csv_filenames = glob.glob('./to-insert/*.csv')
    for csv_filename in csv_filenames:
        try:
            csv_readings, csv_metadata = get_csvs_from_folder(conn, csv_filename)
        except Exception as e:
            log_failure_to_get_csv_readings_from_folder(csv_filename)
            continue
        try:
            insert_csv_readings_into_db(conn, csv_readings, csv_metadata, csv_filename)
        except Exception as e:
            log_failure_to_insert_csv_readings_into_db(csv_filename, csv_metadata)
    conn.close()
```

If you strongly prefer not to use the one continue, I can try to figure out the cases that could arise and deal with them using if statements in the insert...() function.

But I don't think having one continue in main significantly complicates it.

---

Sidenote: 

I use conn in get_csvs_from_folder() to query the database to check if the first reading is already inserted.

Since we're using one conn for the whole script, I added a line to close it at the end of main instead of the end of the log_failure...() and insert...() functions.

carlosparadis, 2018-11-24:
> I would prefer to avoid calling functions that depend on variables we haven't initialized.

What do you mean by this?

matthew-schultz, 2018-11-24:
@carlosparadis 

Okay, I just realized I had an incomplete understanding of how Python determines scope and had to look up some documentation just now.

Apparently Python does not limit scope to __for loops__ or __if statements__. It limits scope by __modules, functions,__ and __class definitions__.
https://stackoverflow.com/questions/3611760/scoping-in-python-for-loops
https://docs.python.org/3/reference/executionmodel.html#naming-and-binding

So I probably should set any variables to the empty string at the beginning of each iteration, so they don't carry over from previous iterations.

That kind of solves the problem I was thinking of that if get_csvs...() catches an exception without calling continue, then insert_csv..() would get called and throw errors since it would try to use the uninitialized variables csv_readings and csv_metadata.

Also anything in ``` if __name__=='__main__':``` is globally scoped in the script since if statements don't limit scope. That would make conn a global variable that we don't need to pass in arguments.

What do you think of this version of main that still uses continue?

```python3
if __name__=='__main__':
    conn = get_db_handler()
    csv_filenames = glob.glob('./to-insert/*.csv')
    for csv_filename in csv_filenames:
        csv_readings = ''
        csv_metadata = ''
        try:
            csv_readings, csv_metadata = get_csvs_from_folder(csv_filename)
        except Exception as e:
            log_failure_to_get_csv_readings_from_folder(csv_filename)
            continue
        try:
            insert_csv_readings_into_db(csv_readings, csv_metadata, csv_filename)
        except Exception as e:
            log_failure_to_insert_csv_readings_into_db(csv_filename, csv_metadata)
    conn.close()
```

I can probably figure out a way to do this without using the continue, but I still prefer this way since it skips calling insert_csv...() when get_csvs...() throws an exception.

carlosparadis, 2018-11-24:
You didn't use the code I pasted with the variable and function renames, so no good! :) Can you copy it and edit to your liking from it so we are sure you didn't miss any of the edits on nomenclature? 

Also, it's a no-no on conn or any other global not being specified as a parameter. It will just mislead everyone on it not being used in the function. 

Notice without the function rename and without conn as a parameter, there is **absolutely no way to guess that function is connecting to the db from main!**. 

I am not too sure on the initialization with empty strings. I have seen this in C, but Python? Do you have references this is good practice?

matthew-schultz, 2018-11-24:
Sorry I forgot to change get_csvs_from_folder in my last comment. Also do want the log functions to mention conn in arguments since they use conn?

```python3

if __name__=='__main__':
    conn = get_db_handler()
    csv_filenames = glob.glob('./to-insert/*.csv')
    for csv_filename in csv_filenames:
        csv_readings = ''
        csv_metadata = ''
        try:
            csv_readings, csv_metadata = get_csv_from_folder_not_in_db(conn, csv_filename)
        except Exception as e:
             log_failure_to_get_csv_readings_from_folder(conn, csv_filename)
             continue
        try:
             insert_csv_readings_into_db(conn, csv_readings, csv_metadata, csv_filename)
        except Exception as e:
             log_failure_to_insert_csv_readings_into_db(conn, csv_filename, csv_metadata)

```

carlosparadis, 2018-11-24:
Yes, the coon on all of them is better, it didn't click on my head! 

I don't need you to convince me on the remainder, just to double check if it is "good practice" and we are not missing anything here. 

Nonetheless, the code looks readable enough for me now, so think of this as the cherry on the top before we call this a day :)

matthew-schultz, 2018-12-01:
@carlosparadis 

I think my main concern is what happens to the values of csv_metadata and csv_readings if get_csv...() throws an exception.

For example:
Say we run the script with 2 files in the HOBO folder, csv 1 and csv 2 without having a continue statement or resetting csv_readings and csv_metadata to None.
We have the first iteration through the for loop in main run successfully so it sets and inserts csv_readings and csv_metadata for csv 1, but say loop 2 throws an exception from get_csv..() 2. __Loop 2 would then call insert_csv...() with csv_readings and csv_metadata still set for file 1__. Loop 2 might then re-insert the readings from loop 1.

However if we instead have a continue statement in get_csv...()'s except block, insert_csv...() would not get called in loop 2.

Does that make sense?

matthew-schultz, 2018-12-02:
@carlosparadis 

I was also thinking about adding one more line to the for loop in main that would be used as the upload timestamp for reading insertions and any database insertion or error logs for a given iteration in the loop like this:

```python3
if __name__=='__main__':
    conn = get_db_handler()
    csv_filenames = glob.glob('./to-insert/*.csv')
    for csv_filename in csv_filenames:
        current_timestamp = pendulum.now('Pacific/Honolulu').format('YYYYMMDD-HHmm')
        try:
            csv_readings, csv_metadata = get_csv_from_folder_not_in_db(conn, csv_filename)
        except Exception as e:
             log_failure_to_get_csv_readings_from_folder(conn, csv_filename)
             continue
        try:
             insert_csv_readings_into_db(conn, csv_readings, csv_metadata, csv_filename)
        except Exception as e:
             log_failure_to_insert_csv_readings_into_db(conn, csv_filename, csv_metadata)

carlosparadis, 2018-12-03:
Where are the timestamps created in the other 2 scripts for webctrl and egauge?

matthew-schultz, 2018-12-03:
@carlosparadis 

In the other two scripts, the timestamp was set at the function level (like insert_readings_into_database(), log_failure...(), etc) instead of the for loop/main level.

However, I would like to eventually go back and change those scripts to match the behavior for HOBO. That should help in the debugging process since any given reading's upload_timestamp should have an identically matching database insertion log upload_timestamps.

carlosparadis, 2018-12-03:
Why don't we leave it inside the function for now for consistency, and then we open an issue to discuss moving this to the main? This will impact how we log things, so for now I prefer we keep the 3 the same way, whichever they may be. 

Explaining someone the log with them in different forms will be confusing, and I think this can be pondered at it's own time!

matthew-schultz, 2019-01-08:
@carlosparadis 

The PR should be here: https://github.com/erdl/sensors/pull/61
None of my 3 PR's include the module-wide classes yet. (Just orm classes corresponding to each table in the db)

matthew-schultz, 2019-03-09:
Notes from 3/8 meeting with Carlos

Error logging during different data_acquisition and database_insertion
---
__data_acquisition step__
- success 
  - insert status="True" rows into error_log for each purpose_id extracted
  - insert into error_log_details
    - error_log.log_id
    - csv_filename, csv_modified_timestamp, earliest_csv_timestamp, latest_csv_timestamp
- failure
  - insert 1 status="False" row for the whole file since purpose_id may not have been matched
  - insert csv_filename and error_log.log_id into error_log_details

__database_insertion step__
- every purpose_id regardless of success, failure, etc
  - insert into error_log_details
    - error_log.log_id
    - csv_filename, csv_modified_timestamp, earliest_csv_timestamp, latest_csv_timestamp
- success
  - for each purpose_id
    - insert status="True" rows into error_log for each purpose_id extracted
- failure
  - for each purpose_id
    - insert status="False" row
- readings were already in database (if new_readings is "False")
  - for each purpose_id
    - insert nothing into status column

sensor_info
---
last_updated_datetime
- check if timestamp of last reading in current csv > last_updated_datetime before updating last_updated_datetime
   - that way even if files are uploaded out of order, last_updated_datetime will have the timestamp of the latest reading for that purpose
   - ex: if one file has a last timestamp in February 2019 and is uploaded before a file with a timestamp in January 2019, the January timestamp won't overwrite the February one.

carlosparadis, 2019-05-21:
Since I accepted the HOBO PR #61 today, I am closing this issue.

closed 2019-05-21 by carlosparadis