erdl/lonoa #62: Sensors project

matthew-schultz, 2018-12-13:
- Decide how to manage adding and removing projects
- Try to keep the project configs/documentation inside the database as well in a single table

matthew-schultz, 2018-12-18:
Since we talked about Java Interfaces in person earlier, I was able to find a Python alternative called Abstract Base Classes.

https://stackoverflow.com/questions/8181576/java-abstract-interface-design-in-python
https://www.python.org/dev/peps/pep-3119/
Abstract Base Class docs - https://docs.python.org/3/library/abc.html
tutorial - https://www.python-course.eu/python3_abstract_classes.php

It looks like we could implement an abstract class with an abstract method like this:

```python3
class AbstractSensor(ABC):
    def __init__(self, sensor_id, conn):
        self.sensor_id=sensor_id
        self.conn=conn
        super().__init__()

    @abstractmethod
    def get_readings_not_in_db_and_insert_into_db(self):
        pass
```

And then create a subclass of the abstract class and instantiate it

```python3
class EgaugeSensor(AbstractSensor):
    sensor_type="EGAUGE"

    def get_readings_not_in_db_and_insert_into_db(self):
        print("do things here")

if __name__ == "__main__":
    egauge_sensor = EgaugeSensor(sensor_id="111", conn="conn")
    egauge_sensor.get_readings_not_in_db_and_insert_into_db()
```

matthew-schultz, 2019-01-01:
Example workflow for adding new projects:

1) Clone project repo from sensors repo
2) Make any necessary updates to newly cloned project repo for new project
3) Add project to project_config table in database with all projects
4) Create new project database based on information entered in project_config
5) Create primary tables in new project database
6) Give crontab account permissions to access and update database and primary database tables
7) Populate database tables with data source info (sensor info, folder info, etc)
8) Create secondary tables in database
9) Populate timestamp tables to determine what time periods to start collecting data over
10) Give crontab account permissions to access and update secondary database tables (timestamp logs, etc)
11) Update crontab with new lines for each subproject

matthew-schultz, 2019-01-02:
1/1/19 Call Notes

Example workflow for adding new projects

1. Create project by cloning sensors repo
2. User creates project database
3. User runs script that creates project database tables (should be easy to use for less technical users)
4. User populates sensor table
5. User runs script that reads sensor table and adds lines to crontab for each sensor type in project
6. Update project files as needed using pull request and git pull
7. Add a new sensor by creating a new subdirectory in sensors project (with its own data acquisition class and ORM class), adding to sensor table, and add to crontab using script

Advantages
- should be relatively quick for new people to learn
- should be relatively easy to update (new sensor types shouldn't need to update any preexisting code)

carlosparadis, 2019-01-08:
# Schema

## readings

|datetime|purpose_id|value|
|-----------|-----------|-----|

## sensor_info

| purpose_id | sensor_type | sensor_part | last_updated_datetime|
|------------|-------------|--------------|------------------------|

## error_log

|log_id|purpose_id|datetime|status|error_type|
|------|-----------|--------|------|-----------|

## error_log_details

|log_id|information| 
|------|-----------|

matthew-schultz, 2019-01-08:
@carlosparadis 
Thank you for putting in the work with @kathrynparadis to outline the schema and update the workflow! It's useful knowing that we still want to use the purpose id. And I'll keep in mind how the error log could help with finding causes of missing data.

I might still need a string column in __sensor_info__ to make api requests to egauge+webctrl and to open the hobo csv folder, but I'm not sure what I should name it, if you want to get input from  @kathrynparadis too.

Maybe "query_string" since one of the current purpose tables uses that?

matthew-schultz, 2019-01-08:
@carlosparadis 

Part of the api request to egauge and webctrl is a string unique to each sensor 
(e.g. ABSPATH:1:#powerscout_frog_2g/kw_avg_phase_b_tn, egauge1, egauge2, etc)

Maybe it would be good to have a long form __sensor_info_details__ table for that kind of stuff, since it varies by sensor type?

carlosparadis, 2019-01-08:
@matthew-schultz sensor_info can't be long_form because it is a very used table by Eileen and Kat. What I forgot was the `sensor_id` column. What you said goes in it. For the HOBO, what goes in is the serial_number of the entire hobo device. 

I updated the comment description to include the project table, and also to contain the `sensor_id` column, as well as extended the explanation of what script uses it on the table explanation. 

This should cover the missing points. Let me know if you have any other question here.

matthew-schultz, 2019-01-08:
@carlosparadis Got it! Yeah, that way they can easily update ```sensor_id``` on their own if they need to.

I will probably need one more table for the username and password used to access the webctrl api; currently it looks like this

webctrl_user
---
user_id | username | password
--- | --- |---

carlosparadis, 2019-01-08:
call it api_authentication instead of webctrl user since in the future we may have more, columns looks fine. i guess thats it!

matthew-schultz, 2019-01-09:
@carlosparadis 

One more thing regarding creating and connecting to the database:

Is it okay if I assume that a given project's folder name will be the same as its database name?
That way, a project's scripts always refer to its top level folder when connecting to the database.

Ex:
If we want to make a new project "frog-1", we clone the sensors repo and name the cloned folder "frog-1". We create a database "frog-1" and the ```api_.py``` scripts read the project folder name to find the name of the database to connect to.

carlosparadis, 2019-01-09:
In general, you want to always ask yourself, "how will people know of this assumption?" and "what would take to avoid such assumption?". 

Given I just mentioned the creation of a project table, what you can just do is **add another column, namely `database_name`** . When your `init_database.py` creates the tables, it needs to take as input the database name where it will do that. It can then store the name in this value for further reference.

matthew-schultz, 2019-01-09:
@carlosparadis 

> In general, you want to always ask yourself, "how will people know of this assumption?" and "what would take to avoid such assumption?".

Maybe a better way for me to put it is that I want to establish a convention instead of make an assumption.

I want to establish a convention for a script to obtain the database name for a given project without connecting to that project's database, since the process of connecting to a given database requires its name.

If we use a project called ```frog-1``` as an example, we need to get the string ```frog-1``` from an external source before connect to the database ```frog-1```.

carlosparadis, 2019-01-09:
Just store in the database as I mentioned. Once we are all gone, this will be noted somewhere, along with all our conventions, and eventually forgotten. It just doesn't work long-term due to the nature of the lab.

matthew-schultz, 2019-01-09:
@carlosparadis 

If it's stored in a project database, then how do I make the initial connection to that project's database?

carlosparadis, 2019-01-09:
I mentioned in the previous comment:

> Given I just mentioned the creation of a project table, what you can just do is add another column, namely database_name . **When your init_database.py creates the tables, it needs to take as input the database name where it will do that. It can then store the name in this column for further reference.**

carlosparadis, 2019-01-09:
@ryantanaka I can't find your diagram where you had some arrows show the symbolic link in the **tmp** folder of scrape util's to var log. Could you upload it to the scrape-util wiki and also here?

@matthew-schultz If he finds it, you will see what kind of madness can spring out of conventions.

ryantanaka, 2019-01-09:
> @ryantanaka when you can, please take a look too!

Regarding the workflow:
- Just wondering, `init_database.py` and `init_crontab.py` are not done yet right? 
- I think I am missing something here, can you explain this one with a tiny example:

> If you're adding a new type of sensor to the table, run the init_crontab.py script. Otherwise, the already existing scripts will automatically start downloading data for it!

The workflow stuff looks good to me. 

Regarding the schema:

- The only feedback I have here is that for certain columns, it might be useful to use [enumerations](https://www.postgresql.org/docs/9.1/datatype-enum.html) that we can create in postgres. For example, for `sensor_type` if we only have the values egauge, webctrl, and hobo, then enum would be a good use here. This can always be extended if we have more sensor types later.  The same can be said for status, if we don't use a bool here, then the enums "success" and "failure" would be good. Currently, the purpose tables are using varchars for everything (see below).. which is totally fine, but maybe things are a little "cleaner" if we use enums, idk. 

```
frog_uhm=> \d+ purpose
                                     Table "public.purpose"
      Column       |          Type          | Modifiers | Storage  | Stats target | Description
-------------------+------------------------+-----------+----------+--------------+-------------
 purpose_id        | integer                | not null  | plain    |              |
 building          | character varying(50)  |           | extended |              |
 variable_name     | character varying(50)  |           | extended |              |
 unit              | character varying(20)  |           | extended |              |
 type              | character varying(50)  |           | extended |              |
 appliance         | character varying(30)  |           | extended |              |
 room              | character varying(30)  |           | extended |              |
 surface           | character varying(50)  |           | extended |              |
 sample_resolution | character varying(20)  |           | extended |              |
 query_string      | character varying(255) |           | extended |              |
 note              | character varying(255) |           | extended |              |
Indexes:
    "purpose_pkey" PRIMARY KEY, btree (purpose_id)
Referenced by:
    TABLE "reading_outliers" CONSTRAINT "purpose_id_fk" FOREIGN KEY (purpose_id) REFERENCES purpose(purpose_id)
    TABLE "reading" CONSTRAINT "purpose_id_fkey" FOREIGN KEY (purpose_id) REFERENCES purpose(purpose_id)
    TABLE "reading_temp" CONSTRAINT "reading_copy_purpose_id_fkey" FOREIGN KEY (purpose_id) REFERENCES purpose(purpose_id)
    TABLE "egauge_5min_avg_trailing" CONSTRAINT "reading_copy_purpose_id_fkey1" FOREIGN KEY (purpose_id) REFERENCES purpose(purpose_id)
    TABLE "kat.reading_copy" CONSTRAINT "reading_purpose_id_fkey" FOREIGN KEY (purpose_id) REFERENCES purpose(purpose_id)
```

> @ryantanaka I can't find your diagram where you had some arrows across directory trees on the server showing the symbolic link in the tmp folder of scrape util's to var log. Could you upload it to the scrape-util wiki and also here?

@carlosparadis, I think this is the one you were looking for :)

[diagram](https://github.com/erdl/legacy-scrape-util/wiki/tmp-project-directory-explained
)

carlosparadis, 2019-01-10:
```
1. Just wondering, init_database.py and init_crontab.py are not done yet right?
2. I think I am missing something here, can you explain this one with a tiny example:

>If you're adding a new type of sensor to the table, run the init_crontab.py script. Otherwise, the already existing scripts will automatically start downloading data for it!
```
1. Correct, they are just a concept at this point.
2. The sensor_info/purpose table currently maintained by hand by Eileen already contains the sensor_ids (e.g. "ABSPATH:1:#powerscout_frog_2g/kw_avg_phase_b_tn", "egauge1", "egauge2" etc) to purpose_id mapping. So, we can have the scripts (egauge_api, webctrl_api, hobo_api) use it, instead of a config file outside repeating the same information (scrape-utils). 

The missing part here, is who will keep running the scripts (egauge_api, webctrl_api, hobo_api, etc_api), by adding them to the crontab. To keep things clean and documented, this would be done by the init_crontab.py. It is basically a script that checks the sensor_info table `sensor_type` column to know which *_api.py are required, and then add them to the crontab. 

The idea is that, if Eileen or Kat later wants to also remove all egauge sensors from the table, it makes no sense to have a crontab of egauge_api.py running without any sensor to collect. It's clock waste. In this case, re-running (manually), init_crontab.py, would "refresh" the crontab to remove the egauge_api.py. 

Notice that, it is the *_api.py scripts that know how to handle 1 or more sensors of it's kind (generalizing this would lead to problems since each sensor has its own way to collect 1 or more sensors). The init_crontab.py just cares to add the right ones to the crontab.

Finally, notice that even knowing which _api scripts to run, you still need to know where they are located, i.e., you need the filepath of the project folder in the server to be known by the init_crontab.py script. This is added by the `init_database.py` script when the tables are created, since the file resides within whatever is the project folder.

Finally, the scripts also need to know which database to look for to load and store data. This is what I will discuss with Matt at 4 PM today on call. He is proposing convention using the folder project name on the server, I am opposing the idea and suggesting we just store it at the project table, which already stores its filepath, and have it passed as a parameter to the `init_database.py` on the first and only time it is executed. 

--------

@ryantanaka Yes, that's the image I was after of one convention Forrest did and we spent hours making sense of it. Thanks for pointing out!

matthew-schultz, 2019-01-12:
> it might be useful to use enumerations that we can create in postgres. For example, for sensor_type if we only have the values egauge, webctrl, and hobo, then enum would be a good use here. This can always be extended if we have more sensor types later. The same can be said for status

@ryantanaka Thank you for mentioning enumerations! I will look into using that datatype where we want to store a limited variety of strings.

@carlosparadis I'm going to update your schema https://github.com/erdl/sensors/issues/62#issuecomment-452172935 so it includes 
__sensor_info__'s new columns (sensor_id and is_active) and the __api_authentication__ table,
if that sounds okay.

matthew-schultz, 2019-01-15:
@carlosparadis 

I'm going to update the schema to include this update to __error_log__ you talked about here https://github.com/erdl/sensors/issues/64#issuecomment-452169152

> After iterating this with @kathrynparadis , I feel the order of the rows is too subtle to distinguish, so let's go with one more column called "pipeline_stage" which has as values "data_acquisition" or "database_insertion". This will let anyone distinguish the 2 rows easily.

matthew-schultz, 2019-01-15:
@carlosparadis 

I think we need 2 information columns instead of just one for the __error_log_details__ table

Is it okay if I split the information column into two like this? (And if so, do the new column names make sense?)

log_id | information_name | information_value
-- | -- | --

carlosparadis, 2019-01-15:
Yes, that's fine.

Don't jump into enumerations yet, who knows if Tableau can read that when it doesn't even know timestamps with timezone. For now, just go with strings.

matthew-schultz, 2019-01-23:
@carlosparadis 

Remember how we were talking about reading the database name from a text file? (since we can't connect to the database using a value retrieved from the database)

Does it sound okay if I just add code like that to the api_ file's get_db_handler() function that connects to the database?

```python3

    db_url_config = "/db_url_config.txt"
    with open(db_url_config, "r") as db_url_config_file:
        db_url = db_url_config_file.read()

carlosparadis, 2019-02-01:
that is fine, just call it "config.txt", n case we ever need to add anything else to it instead of the DB. Inside, I think `db = <dbname>`.

matthew-schultz, 2019-02-02:
@carlosparadis 

What do you think about creating a separate function __get_db_url()__ in the api scripts, since I will probably also need the db url for functions that create and drop all the database tables for testing?

I might have to add one more line to the main block that calls that function, but the whole process of getting the config file path, reading it, and parsing the db field is about 6 lines, so it'd be useful to have a function to call instead of copy-pasting those lines.

Also, if we find a better way to get the database url, then we have it encapsulated in a function and can just update that function without changing other code.

```python3
config_path = str(Path(os.path.dirname(os.path.realpath(__file__))).parent.parent) + "/config.txt"
with open(config_path, "r") as file:
    #prepend '[DEFAULT]\n' since ConfigParser requires section headers in config files
    config_string = '[DEFAULT]\n' + file.read()
config = configparser.ConfigParser()
config.read_string(config_string)
db_url = config['DEFAULT']['db']
```

matthew-schultz, 2019-02-13:
Notes from 2/12 meeting with Carlos
- Keep the db_url code duplicated for now; will decide to create a function like __get_db_url()__ later
- error_log table probably doesn't need to have sensor_id, but check with Eileen to make sure

matthew-schultz, 2019-03-09:
Notes from 3/8 meeting with Carlos

Update sensor_info table
---
- add columns
  - units - units of a purpose's readings
  - data_sensor_info_mapping
    - used to match a reading with its purpose_id (instead of sensor_part)
    - matches full column name in raw data (egauge api data, hobo csv's, etc)
    - user manually enters data into column
    - simpler for users to understand without having to look at code to see how column name was parsed

init_database.py
---
- takes database name and crontab user as arguments
- print error message with proper argument example (e.g. run using command "init_database.py some_database_name some_user")
- creates database
- creates tables based on orm file
- creates config.txt file with database name and crontab user name

user associated with crontab that runs scripts ("daemon")
---
- add name of user to config.txt
- research
  - creating user with limited permissions that will run crontab
  - figure out how to assign crontab to user (can contact Ryan), update script

carlosparadis, 2019-05-21:
Ah, so much discussion to get to our current workflow :') I'm glad today paid off the effort. As I accepted the PR of egauge, webctrl and HOBO today, which implements this concept, I am closing this issue. Init_database.py PR #68 is still pending as of now, but should be done relatively soon anyway.

closed 2019-05-21 by carlosparadis