erdl/lonoa #63: Create classes for each sensor

matthew-schultz, 2018-12-20:
Create a class for each sensor type that:
- has a run() function that will get data for all sensors of that type
- includes all attributes needed for that run() function (list of sensor or sensor ids)
- uses the same tables (Reading, Database_Insertion_Timestamp, Sensors) as the other sensor classes
- will run on test server from crontab
- try to keep in mind how we will add new projects

matthew-schultz, 2019-01-02:
1/1/19 Call Notes

- Update data acquisition classes to obtain readings from multiple sensors
- Avoid using a Purpose table for now; if a given sensor (e.g. hobo) has multiple readings per timestamp, insert all readings
- Use tables with minimum amount of necessary columns
  - sensor_config table + long form table (if needed)
    - id, sensor_type, sensor_path
  - reading table
    - id, sensor_id, value, unit
  - database_insertion_timestamp table + long form table (if needed)
    - id, sensor_id, sensor_type, status, upload_timestamp, latest_timestamp

Next objective - update api_egauge and orm_egauge and make a PR

matthew-schultz, 2019-01-02:
@carlosparadis 

I think I'm pretty close to having the EguageAPI and EgaugeORM classes ready; should I go ahead and do a pull request? Or do you want to check over anything else first like commit messages, table structure, etc?

carlosparadis, 2019-05-21:
For now, I am retiring this effort, please keep the code on your computer just in case, but after going back and forth on scrape-util over the last few months, and in light of future students not necessarily not knowing OO, we decided to abide by KISS.

closed 2019-05-21 by carlosparadis