erdl/lonoa #64: Update timestamp log table to work with egauge, webctrl, and hobo scripts

matthew-schultz, 2019-01-05:
Per 1/4/19 ERDL meeting, update ORM table classes to specifications and eguage, webctrl, and hobo scripts to use the same tables.

Insert 2 rows into timestamp table per sensor during a script run
Store extra data in long timestamp table as necessary for each sensor (E.g. hobos)
For every insertion into timestamp table, check if we need to insert anything into long table

matthew-schultz, 2019-01-08:
@carlosparadis 

I was able to make some updates to the egauge classes and tables; does this match up with what we were talking about for the __database_insertion_timestamp__ table?

 id | sensor_id | is_success |     download_timestamp     |      upload_timestamp      
--- | --- | --- | --- | ---
 1 | egauge1 | True          | 2019-01-01 12:20:20.782214 | 
2 | egauge1 | True          |                            | 2019-01-01 12:20:00.25771
  3 | egauge2 | True          | 2019-01-01 12:20:21.336705 | 
  4 | egauge2 | True          |                            | 2019-01-01 12:20:21.846321

While __sensor_config__ would look something like this:

id |                    sensor_id                    |  type   |    checkpoint_timestamp    
--- | --- | --- | ---
  1 | egauge1                                       | EGAUGE  | 2019-01-01 12:19:00
  2 | egauge2                                       | EGAUGE  | 2019-01-01 12:19:00
  3 | ABSPATH:1:#powerscout_frog_2g/kw_avg_phase_b_tn | WEBCTRL | 2019-01-01 10:00:18.538879
4 | ./hobo-readings-path/ | HOBO | 2019-01-01 10:00:18.538879

carlosparadis, 2019-01-08:
@matthew-schultz The first table should just have one column for datetime. Otherwise, it will alternate NAs, and as I said in the meeting with @ryantanaka, this is bad design. 

After iterating this with @kathrynparadis , I feel the order of the rows is too subtle to distinguish, so let's go with one more column called "pipeline_stage" which has as values "data_acquisition" or "database_insertion". This will let anyone distinguish the 2 rows easily. 

As for the second table, and after iterating again with @kathrynparadis , we decided to use purposes ids on this design as well. As such, you should remove the columns `sensor_id`, `type`, and add a new column, called `purpose_id`.

carlosparadis, 2019-05-21:
Very nice, I suppose this issue is done per what I saw earlier on sea-local, and given I accepted the PRs for egauge, webctrl and HOBO. Feel free to re-open this issue and call my attention otherwise.

closed 2019-05-21 by carlosparadis