erdl/lonoa #71: Refactor ORM files in Egauge, Webctrl, and HOBO

carlosparadis, 2019-05-22:
On the current code, there are 2 classes on the ORM that are really enumerations and refer to columns of the other tables. I mentioned them in the PRs reviews. Is there a way we can just put them back in the classes they belong?

matthew-schultz, 2019-05-25:
Notes from 5/24/19

- Prioritize finishing this task within the next week
- Try to use an enum within the table class instead of using it's own separate class

matthew-schultz, 2019-05-28:
Notes 
- converted 2 orm classes to inherit from python enum class
   - converted ```PipelineStage``` to ```PipelineStageEnum```
   - converted ```SensorType``` to ```SensorTypeEnum```
- moved enum classes into table classes
   - moved ```PipelineStageEnum``` into ```ErrorLog```
   - moved ```SensorTypeEnum``` into ```SensorInfo```
- updated __api__/__extract__ scripts to refer to ```SensorInfo.SensorTypeEnum``` and ```ErrorLog.PipelineStageEnum```
- updated __init_crontab__ to access string stored ```SensorTypeEnum.value```

carlosparadis, 2019-05-30:
@matthew-schultz I am not positive we need a class inside a class. The question is, do we **really need** a Enum class just to represent an enumeration in Python? Seems way too much overhead.

matthew-schultz, 2019-05-30:
@carlosparadis
Yeah, the way python3 implements enumerations is through classes 
But I'm not sure if that makes it easier for us to maintain the code in the long term 😬

https://docs.python.org/3/library/enum.html
> Enumerations are created using the class syntax, which makes them easy to read and write. An alternative creation method is described in Functional API. To define an enumeration, subclass Enum as follows:
```python3
from enum import Enum
class Color(Enum):
...     RED = 1
...     GREEN = 2
...     BLUE = 3
... 
```

Would it be better just to use strings for ```sensor_info.sensor_type``` and ```error_log.pipeline_stage```  instead of using enum classes?

I was thinking classes would be a good way to have those values in one place in case we needed to update them, but I'm not sure if that outweighs having to know how to work with the enum python class and the enum data type in postgres.

carlosparadis, 2019-05-30:
@matthew-schultz So you can send a PR #73, regardless of what happens in this issue, am I understanding correct [only the last commit](https://github.com/erdl/sensors/pull/81/commits/001ad4a5c13dc0dbe37f6fc0e4178238a0df4315) modifies init_crontab.py?  

If so, roll this branch back before the commit before sending PR to issue #73. 

Edit: You already sent the PR, but not a problem. You can just adjust the commits on both branches to reflect what I proposed.

matthew-schultz, 2019-05-30:
@carlosparadis Yes, that's correct; only the last commit in this branch's PR #81 modifies init_crontab. And I've tested both PR branches on sea-local

I'll go ahead and update it by 
- removing the commit with the changes to init_crontab 
- updating the other PR to include only commits updating init_crontab

matthew-schultz, 2019-06-01:
@carlosparadis 

I've updated this issue's PR https://github.com/erdl/sensors/pull/81 to include the latest updates of erdl/master and along with the init_crontab enum usage commit, so it should be ready now.

I also tested how the enum type worked in postgres to make sure I was able to add new values to it and drop it.

Adding new value 'raspberry_pi' to sensortypeenum
```ALTER TYPE sensortypeenum ADD VALUE 'raspberry_pi';```

Dropping that enum
```DROP TYPE sensortypeenum;```

matthew-schultz, 2019-07-24:
Closing this issue since PR #81 was merged

closed 2019-07-24 by matthew-schultz