erdl/lonoa #72: Add units to the readings table

carlosparadis, 2019-05-22:
This is important and a needed feature if we are to use this in production. Egauge, Hobo and Webctrl need to also check every time they run the unit column and insert it for all the readings.

Note this also means that if at any point Eileen or Kat change it in the database, the next time the script runs the reading **from then onwards** will be inserted with the new units.

matthew-schultz, 2019-05-22:
Okay, I'm going to work on this next if that sounds good.

Will try to get the pull request in by the end of the day, since it shouldn't be too difficult

matthew-schultz, 2019-05-23:
Just want to make sure: we want to use the plural form ```units``` in the __readings__ table and not the singular ```unit``` right?

In __sensor_info__, I use ```unit``` but I can go ahead and change it to ```units``` so it's consistent

carlosparadis, 2019-05-23:
Yes, currently in production is called "units" so we will want to stick with that.

matthew-schultz, 2019-05-23:
Sorry, going to push to my branch one more time before I make the PR to make sure no extra commits are added to master.

(Earlier on the pull request page, I noticed extra commits from #65 and #66; 
I think they were do to an error when I was editing this issue's commit messages.
Rebased this branch to master so they should no longer be there.)

matthew-schultz, 2019-05-25:
Closing this issue since the pull request was accepted

closed 2019-05-25 by matthew-schultz