erdl/lonoa #73: Fix crontab update logic

carlosparadis, 2019-05-22:
This is one more thing we need if we are to roll to production. Instead of wiping clean the file, you should have your python code actually code parse it and load the file as a dictionary in Python. 

You can then avoid deleting things that are unrelated to sensors from the cron file. I think it is safe toassume that whatever the path of the cronjob is, it will have /sensors/ in it as every project git clone the sensors project (or whatever we call it in the future). So you can hardcode that assumption, and we should add this to the wiki so it is not forgotten. If you have a better idea, do share.

matthew-schultz, 2019-05-25:
Was thinking that I would try to update the code to 
- get a list of unique ```sensor_types``` from ```sensor_info```
- build path strings from ```project.project_folder_path``` + ```sensor_type``` + "/script"
   - ex "/home/locked_user/sensors/" + "hobo" + "/script"
- select all active cronjobs that contain those strings
- delete those selected jobs

Then write the new jobs like it currently does

matthew-schultz, 2019-05-30:
@carlosparadis, I have this update working on sea-local; currently, it:
- reads project_path from the database
- searches locked_user's crontab for any jobs containing that path
- removes those jobs
   - does not remove any commented out jobs (starting with '#') and any jobs that don't contain the path
- adds crontab jobs for active sensor_types like it previously did

Is it okay if add this update to the existing PR #81 since it also updates init_crontab.py?

carlosparadis, 2019-05-30:
@matthew-schultz This should go on its own PR. From the title of the existing PR, they seem disjoint concerns. 

Provided I accept #81 before your upcoming merge, Github should not have any issues merging it after (it will be a fast-forward merge after all).

Go ahead and send it, and I will take a look at the logic you just mentioned. At a glance, it seems fine.

matthew-schultz, 2019-07-24:
Closing this issue since PR #82 was merged

closed 2019-07-24 by matthew-schultz