erdl/lonoa #76: Rename columns of sensor_info to indicate which are used by the script

carlosparadis, 2019-05-24:
This is high since there is currently no way for Eileen and others to see when using the table which they can "mess with" and which they can't. 

I am suggesting just renaming the columns with the current name plus `_c` to their end. For example:

sensor_type -> sensor_type_c
purpose_id -> purpose_id_c

etc

Let's just do this on `sensor_info` table. Can you confirm this is the only table in the database that **reads** to do something?

matthew-schultz, 2019-05-24:
The tables ```api_authentication```, ```readings```, and ```project``` are also read by scripts
- api_webctrl reads ```api_authentication``` to get the ```username``` and ```password``` to access webctrl
- extract_hobo queries ```readings``` to see if a hobo csv's first and latest readings are already inserted (it only does this for one of the csv's ```purpose_id```s since all columns in a csv are in one commit)
- init_crontab reads ```project.project_folder_path``` to write the full path of crontab commands 
   - we can do this without querying the database, but in that case we might have to be careful about the working directory when we call this script

Just want to make sure: when we say "mess with" would that include things like Eileen renaming existing columns? 
Or just changing values in a specific row? like how we've discussed certain columns would be like a scratch pad for Eileen to record things

carlosparadis, 2019-05-24:
Anything goes on how she can possibly edit. I will have @kathrynparadis and @eileenpeppard chime in if they think these tables need any indicator or we can just write down in the wiki. If you see them in office and they don't respond here, ask them directly and put a note here. 

While this is flag as high priority since it is needed for usage, compared to other things this has no rush to be worked on for now.

carlosparadis, 2019-08-18:
In order to keep backward compatibility to the views, we can't rename the columns to indicate which can be changed and which can not. This information will be on the wiki for checking when necessary.