erdl/lonoa #77: View compatibility, column naming and purpose table in production uhm_frog

carlosparadis, 2019-05-24:
To move this code to production, we will need to get the purpose table from uhm_frog (similar to sensor_info) in Production, and "adapt" to our current code, as Eileen views rely on the column names to run.

Let's worry about this after we fiddle with the rest of pending issues. I will leave it as high anyway because this can't be used in practice without addressing that.

carlosparadis, 2019-05-24:
Also, Eileen asked to rename  sensor_part -> sensor_type.

I suggest we rename the current sensor_type to script_folder since they map consistently to the sensors project. We can make these changes when we start addressing this compatibility.

matthew-schultz, 2019-06-01:
I'll be working on this issue next

I was able to get a csv of the purpose table from @kathrynparadis 

Next steps
- restructure data for usage in sensor_info table structure
- rename columns in sensor_info
- be mindful of space usage on sea-local, because we've run out of space before
   - can delete readings from before today to save space
- get feedback on new structure of sensor_info
- test on desktop and on sea-local after consideration and feedback

matthew-schultz, 2019-06-06:
| purpose_id | sensor_type | sensor_id | sensor_part | last_updated_datetime| is_active
|------------|-------------|------------ |-------------|------------------------|-------------|

| purpose table | current sensor_info table | target |
| --- | --- | --- |
purpose_id,
building,
variable_name,
unit,
type,
appliance,
room,
surface,
sample_resolution,query_string,note

matthew-schultz, 2019-06-08:
@carlosparadis Should I start trying to run sensors with all the active sensors from the exported __purpose__ table on sea-local?

I was able to convert the current __purpose__ table to __sensor_info__ structure and import it. I've been testing running sensors with those purposes on my desktop vagrant VM and it seems to be running successfully.

Also met with @eileenpeppard today to discuss the __purpose__ table and how its data is used in __sensor_info__

Notes from meeting
---
- We want all the current fields in the __purpose__ table in the new database/ __sensor_info__ table
- There are no hobo sensors that we are actively gathering data for right now; frog buildings just use webctrl and egauge
- For egauge purposes we can use "Usage [kW]" column 
   - Other fields are there for reference just in case HNEI wants them
   - values in "Usage [kW]" are negative when FROG buildings are generating; positive when usage exceeds PV generation
   - Other columns with A, B, C correspond to three loads in the frog building, but we currently just want the combined data
- Probably don't need to get data for 2 purposes where query_string="calculated whole bldg - other measures"
   - those purposes are not currently used

TODO
---
- Check data types of __purpose__ table
   - especially purpose_id 
- Update sensors code for compatibility with __purpose__
   - Rename existing columns in __sensor_info__ where appropriate
      - separate commits for each rename?
   - Add missing columns from __purpose__ table
- Test updates on sea-local

carlosparadis, 2019-06-09:
Yes, go ahead and run all sensors. Just remember to check disk space, and keep tabs on how much it is taking so we don't run out of space again. 

Also, for now just download data starting as of today rather than retro-ing the entire data that already exists.

matthew-schultz, 2019-06-13:
Proposed schema of sensor_info 
---
(includes columns from purpose table)

Column  | Type | Collation | Nullable |  Default | Storage  | Stats target | Description 
| --- | --- | --- | --- | --- | --- | --- | --- |
 purpose_id               | integer                     |           | not null | nextval('sensor_info_purpose_id_seq'::regclass) | plain    |              | 
 query_string             | character varying           |           |          |                                                 | extended |              | 
 data_sensor_info_mapping | character varying           |           |          |                                                 | extended |              | 
 script_folder            | enum?           |           |          |                                                 | extended |              | 
 is_active                | boolean                     |           |          |                                                 | plain    |              | 
 last_updated_datetime    | timestamp without time zone |           |          |                                                 | plain    |              | 
 unit                     | character varying           |           |          |                                                 | extended |              | 
 building                 | character varying           |           |          |                                                 | extended |              | 
 variable_name            | character varying           |           |          |                                                 | extended |              | 
 type                     | character varying           |           |          |                                                 | extended |              | 
 appliance                | character varying           |           |          |                                                 | extended |              | 
 room                     | character varying           |           |          |                                                 | extended |              | 
 surface                  | character varying           |           |          |                                                 | extended |              | 
 sample_resolution        | character varying           |           |          |                                                 | extended |              | 
 note                     | character varying           |           |          |                                                 | extended |              |

matthew-schultz, 2019-06-13:
@carlosparadis I was able to import all the purposes onto sea-local, if @eileenpeppard wants to view the data there some time

matthew-schultz, 2019-06-16:
@carlosparadis 

 > Once we are sure all data is being acquired from the existing sensors, we should create a new issue associated with testing the production database queries of interest, and also put it on high priority.

Just want to make sure, for the acquiring all data from existing sensors step: was that just having the exported production purposes running on the sensors project on sea-local?

Or was there further verification that we wanted to do? (Since I have those purposes running on sea-local now).

carlosparadis, 2019-06-16:
There are actual Queries and Views that you should ask @eileenpeppard which are used in production. You should create an issue containing the name of the views and their location in the existing database, version them somewhere here (make sure there is no private information hardcoded in them like locations).

matthew-schultz, 2019-07-31:
Closing this issue since PR #102 implemented table and column compatibility updates and views were tested in #94

closed 2019-07-31 by matthew-schultz