erdl/lonoa #93: api_webctrl.py currently does not handle duplicate readings

matthew-schultz, 2019-06-15:
Summary
---
- multiple purpose_ids on webctrl had duplicate readings around 12:38 PM today (6/14)
- api_webctrl.py doesn't handle duplicate readings
- going to skip to 12:40 PM to avoid those readings

---

While looking through the sensor_info table, I noticed that 48 webctrl sensors had not run successfully since 12:35 PM today (6/14). 

I checked the error log file and noticed a lot of IntegrityErrors being thrown, which suggested duplicate readings being entered that violated the purpose_id, datetime key.
Looking up one of the sensors on the webctrl server I noticed that it had duplicate readings a little after 12:35 PM 

| | |
| --- | --- |
| Fri Jun 14 2019 12:30:00  |   52.78
| Fri Jun 14 2019 12:35:00  |   51.51
| Fri Jun 14 2019 12:38:23  |   0
| Fri Jun 14 2019 12:38:23  |   0
| Fri Jun 14 2019 12:38:23  |   0
| Fri Jun 14 2019 12:40:00  |   53.51
| Fri Jun 14 2019 12:45:00  |   51.55

I sampled a few other purposes and they also had duplicate readings in the 12:35 to 12:40 range.
For now, I'm going to skip over the duplicate times and just update last_updated_datetime to Fri Jun 14 12:40:00, but it would probably be good to handle this in api_webctrl.py in the future

Possible solution for api_webctrl.py to handle duplicates
---
- convert json to list of timestamps
- check for duplicate timestamps
- if duplicate - remove from json

carlosparadis, 2019-06-15:
@ryantanaka Correct me if I am wrong, but isn't 6/14 the date where scrape-utils had issues to download data as well? Seems to me we found our true culprit here. 

@matthew-schultz Are you saying Webctrl is storing duplicated rows behind the API?

matthew-schultz, 2019-06-15:
@carlosparadis yeah, that's correct; Webctrl seems to be storing duplicate rows.

I'm talking about the latter: api_webctrl.py not being able to insert readings if the API returns data with duplicate times.

carlosparadis, 2019-06-15:
Just for completeness sake, if webctrl had no duplicates, but for some reason the data already existed in the database, would it work? (Thats the former case).

matthew-schultz, 2019-06-15:
No, currently it would not.

If there are readings after __sensor_info.last_updated_datetime__ already in the database for a given purpose_id, it would throw an error on insertion and not insert any readings for that run of api_webctrl.py for that purpose_id.

matthew-schultz, 2019-06-15:
@carlosparadis 

Should I start working on a branch updating api_webctrl so it removes duplicates from the data, before attempting insertion?

I can probably achieve that like extract_hobo does by converting the data to a pandas dataframe and running drop_duplicates() or doing something similar with a sorted list.

I'm not sure about a good way to filter out readings already in the database if last_updated_datetime falls behind, unless we want to query the latest readings' datetime for a purpose and only insert readings after that.

carlosparadis, 2019-06-15:
@matthew-schultz Not yet. Although this is very important, our top priority now is #77. Once we are sure all data is being acquired from the existing sensors, we should create a new issue associated with testing the production database queries of interest, and also put it on high priority. 

If the queries are not guaranteed to move over, the entire project falls apart, since Eileen won't be able to use it, so it's all for nothing.

matthew-schultz, 2019-11-19:
Closing this issue since PR #111 was accepted

closed 2019-11-19 by matthew-schultz