erdl/lonoa #94: Obtain and test views from production database

matthew-schultz, 2019-06-20:
Developing this issue was previously discussed in issues #77 and #93

Next steps
---
- get views from production database (note that production database is currently messy)
- start with @kathrynparadis's views, move on to public after it's been cleaned 
- test for compatibility with current database
- store in local file for now
   - don't version yet, since it might be more appropriate to store in database repo
- try to keep trace and keep track of view dependencies in file, if it doesn't take too long
- decide on a plan for columns that are used by lonoa and are in both purpose and sensor_info
   - renaming columns with '_c' might break views

matthew-schultz, 2019-06-21:
Compare values of shared columns between scrape-util.sensor_info and lonoa.sensor_info to make sure they match
- export values from those columns as files
- diff files to make sure those values match

kathrynparadis, 2019-06-21:
(NOT DONE)

kat schema views:

- dashboard-fans_and_outdoor_lights:
Grabs the outdoor lights 
- dashboard-hvac

-

matthew-schultz, 2019-06-21:
I was able to compare exports of the shared columns in the scrape-util.sensor_info and lonoa.sensor_info tables and they matched.

First, I ran these commands in psql on production and development servers to export the data
```\copy (select purpose_id, query_string, unit from sensor_info order by purpose_id asc) to lonoa_sensor_info.csv header csv```

```\copy (select purpose_id, query_string, unit from sensor_info order by purpose_id asc) to scrape-util_sensor_info.csv header csv ```

Used scp to copy lonoa_sensor_info.csv onto the server and then ran:
 ```diff scrape-util_sensor_info.csv lonoa_sensor_info.csv```

I'm planning on updating lonoa's reading table next, so it matches production for testing the views.

carlosparadis, 2019-06-21:
> I'm planning on updating lonoa's reading table next, so it matches production for testing the views.

As a reminder, you will need to verify the space left available as well as how much the readings table takes on disk space before doing that.

matthew-schultz, 2019-06-21:
Approximate space left in sea-local: 1.3 Terabytes
Approximate space used by readings table on sea-local: 32 MB

---
It looks like there's about 1.3 Terabytes left for the directory mounted on root, when I ran ```df -h``` on sea-local

I've been trying to run that command for the past week when I come in, just to see how fast lonoa uses up, and it looks like it's used up around 1 GB over the past two weeks.

Looks like the readings table is about 32 MB with roughly 288585 rows going from 2019-06-12 17:26:00 to 2019-06-20 17:52:00

```
raichu=# select pg_total_relation_size('readings');
 pg_total_relation_size 
------------------------
               32112640
(1 row)```

kathrynparadis, 2019-06-21:
My previously started comment on view summaries is now finished.

matthew-schultz, 2019-06-22:
I've started working on testing view __dashboard-fans_and_outdoor_lights__

- the view seemed to run successfully on lonoa on my vagrant VM and displayed data
- created a file dashboard-fans_and_outdoor_lights.sql with code and a description along with other information

I might also try to compare the data from this view from both lonoa and scrape-util just to make sure they match.

carlosparadis, 2019-06-22:
ok, looks like space is safe.

@matthew-schultz for testing views, you should strictly test them on the test server to ensure what we move to production from there also works.

carlosparadis, 2019-06-22:
@matthew-schultz is that space on local or production? we need the space for both

matthew-schultz, 2019-06-22:
> for testing views, you should strictly test them on the local server to ensure what we move to production from there also works.

Okay, I'll start running this branch on sea-local then. And will follow the method you mentioned for testing, too.

> is that space on local or production? we need the space for both

That's just on sea-local. I'll check the storage space and reading table space on production right now.

matthew-schultz, 2019-06-22:
Approximate space left in production: 771 Gigabytes
Approximate space used by readings table on production: 3.7 Gigabytes

```
frog_uhm=> select pg_total_relation_size('reading');
 pg_total_relation_size 
------------------------
             3746488320
(1 row)
```

carlosparadis, 2019-06-22:
Sounds like we're safe on both ends!

matthew-schultz, 2019-06-22:
While testing some views I noticed some 'Solar-1' appliance readings were ```-0``` on local and ```0``` on production

Not sure if that'll negatively affect any queries in the future, so I wanted to take note of that for now.

---
```
raichu=> select * from reading
where purpose_id = ANY (ARRAY[38, 50, 94, 107]) 
and datetime > (timestamp ' 2019-06-21 19:15:00')
and datetime < (timestamp '2019-06-21 19:30:00');
      datetime       | purpose_id | reading | units 
---------------------+------------+---------+-------
 2019-06-21 19:20:00 |         38 |      -0 | kW
 2019-06-21 19:25:00 |         38 |      -0 | kW
 2019-06-21 19:20:00 |         50 |      -0 | kW
 2019-06-21 19:25:00 |         50 |      -0 | kW
 2019-06-21 19:20:00 |         94 |      -0 | kW
 2019-06-21 19:25:00 |         94 |      -0 | kW
 2019-06-21 19:20:00 |        107 |      -0 | kW
 2019-06-21 19:25:00 |        107 |      -0 | kW
(8 rows)
```

```
frog_uhm=> select * from reading
frog_uhm-> where purpose_id = ANY (ARRAY[38, 50, 94, 107]) 
frog_uhm-> and datetime > (timestamp ' 2019-06-21 19:15:00')
frog_uhm-> and datetime < (timestamp '2019-06-21 19:30:00');
      datetime       | purpose_id | units | reading |    upload_timestamp    
---------------------+------------+-------+---------+------------------------
 2019-06-21 19:20:00 |         38 | kW    |       0 | 2019-06-21 19:21:23.32
 2019-06-21 19:25:00 |         38 | kW    |       0 | 2019-06-21 19:41:24.38
 2019-06-21 19:20:00 |         50 | kW    |       0 | 2019-06-21 19:21:23.32
 2019-06-21 19:25:00 |         50 | kW    |       0 | 2019-06-21 19:41:24.38
 2019-06-21 19:20:00 |         94 | kW    |       0 | 2019-06-21 19:21:23.32
 2019-06-21 19:25:00 |         94 | kW    |       0 | 2019-06-21 19:41:24.38
 2019-06-21 19:20:00 |        107 | kW    |       0 | 2019-06-21 19:21:23.32
 2019-06-21 19:25:00 |        107 | kW    |       0 | 2019-06-21 19:41:24.38
(8 rows)
```

carlosparadis, 2019-06-22:
@matthew-schultz may want to check with @eileenpeppard if she has any idea on this -0!

matthew-schultz, 2019-06-25:
I'm also testing the public views __eguage_5min_avg_view2__ (sic) and __egauge_5min_avg_view1__ , since some of @kathrynparadis's views depend on them. Let me know if these descriptions make sense or not.

- egauge_5min_avg_view1
Assigns a five minute incremented datetime to each "Whole building net egauge" appliance reading (:00, :05, :10, etc)
[used by eguage_5min_avg_view2]

- eguage_5min_avg_view2
Averages "Whole building net egauge" appliance readings in 5 minute increments
[used by dashboard-pv-pos-whole-building-net, dashboard-readings-neg, dashboard-whole-building-net]

matthew-schultz, 2019-06-26:
@carlosparadis I was able to successfully run all of kat's views and store them in sql files; also talked with @eileenpeppard today and I don't think she's seen readings with ```-0``` in the database before either.

Some of the views had references to kat's schema so I removed those references from the query file so they would run on sea-local, but I can add them back if we prefer.
- dashboard-whole-building-use, dashboard-plugloads, dashboard-union_individual, dashboard-union_of_views

Some views with column values that were manually set to strings also needed to be typecast as varchar to run.
- dashboard-whole-building-use, dashboard-union_of_views

One view also returned a slightly different value (~.0000000000002 kW); I'm not sure if we want to look more into that, since it seems to be an insignificant amount
- dashboard-plugloads

```
raichu=> select * from "dashboard-plugloads"
raichu-> where datetime < (timestamp '2019-06-19 13:00:00')
raichu-> and datetime > (timestamp '2019-06-19 11:59:59')
raichu-> order by datetime,building,appliance
raichu-> limit 1;
 building | appliance |      datetime       |  Power Avg (kW)   
----------+-----------+---------------------+-------------------
 frog-1   | Plugloads | 2019-06-19 12:00:00 | 0.149999999999999
(1 row)
```
```
frog_uhm=> select * from kat."dashboard-plugloads"
where datetime < (timestamp '2019-06-19 13:00:00')
and datetime > (timestamp '2019-06-19 11:59:59')
order by datetime,building,appliance
limit 1;
 building | appliance |      datetime       |  Power Avg (kW)   
----------+-----------+---------------------+-------------------
 frog-1   | Plugloads | 2019-06-19 12:00:00 | 0.150000000000001
(1 row)

```

I'll start going through the public views in production if that sounds good.

carlosparadis, 2019-06-26:
@matthew-schultz 

> I'm also testing the public views eguage_5min_avg_view2 (sic) and egauge_5min_avg_view1 , since some of @kathrynparadis's views depend on them. Let me know if these descriptions make sense or not.

She will get back to you on this.

------

Based on what you did, would you be able to assign the queries among a) lonoa each sensor folder, b) lonoa root level query folder, and c) database query folder, or would you need help?

Also, what is the list of public queries? Is that posted somewhere?

kathrynparadis, 2019-06-26:
Yes Matt, your summaries for the 2 egauge views are correct.

matthew-schultz, 2019-06-26:
@carlosparadis

> Based on what you did, would you be able to assign the queries among a) lonoa each sensor folder, b) lonoa root level query folder, and c) database query folder.

Kind of... I was able to figure out what sensors each query used, so if we wanted to do just ```a) lonoa each sensor folder``` and ```b) lonoa root level query folder```, I could do that. 

But I'd probably need help distinguishing between queries that would go in ```b) lonoa root level query folder``` or ```c) database query folder```

> Also, what is the list of public queries? Is that posted somewhere?

I just ran a query selecting all views with the public schema to get the list
```
frog_uhm=> select table_name from INFORMATION_SCHEMA.views 
where table_schema='public' order by table_name;
                  table_name                  
----------------------------------------------
 calc-other-inst-power
 conference-data-6mo
 dashbd_indoorenv_sched
 dashbd-indoorenv-sched-testweathercombine
 dashbd_indoorenv_weatherstation
 dashboard-with-indoor-envir
 dates_kevins_meter
 dates-pmv
 dates_reading
 egauge_5min_avg_view1
 eguage_5min_avg_view2
 meter-comparison-1minute
 meter-egauge-minute
 meter-kevin-minute
 meter-kevin-seconds
 model_as_columns_view
 model_data_view
 model_whole_bldg_kwh_view
 power-readings-all
 power-readings-only
 single_line_CO2_temp_weather_ac_fans
 single_line_CO2_temp_weather_ac_runtime_fans
 view-as-columns-all-power
 view-data-comfort-analysis
 view-egauges
 view-indoor-envir
 view-readings-as-columns
 view-schooldays-by-day
 view_weather
 view-whole-bldg-both-recent
 whole_bldg_egauge_vs_wbctrl
(31 rows)
```

carlosparadis, 2019-06-26:
@matthew-schultz Did eileen already removed the views that are useless?

matthew-schultz, 2019-06-26:
@carlosparadis I'm not sure; I'll ask eileen tomorrow

Maybe I'll just test ```view-indoor-envir``` and ```dashboard-with-indoor-envir``` for now, since I think at least those are still used

eileenpeppard, 2019-06-26:
I did delete a bunch of them last week.

eileenpeppard, 2019-06-26:
I deleted a couple more this morning.

matthew-schultz, 2019-06-26:
Thank you, @eileenpeppard 

I'll go ahead and test the public schema views that remain, then

matthew-schultz, 2019-06-28:
I tested a few more public views yesterday; let me know if these descriptions make sense.

calc-other-inst-power
- calculate sum of instantaneous power for each building
[used by NA]

conference-data-6mo
- grab non-egauge Power-avg type readings from 2017-03-22 to 2017-09-21
[used by NA]
- note: line using reading.upload_timestamp was temporarily commented out for testing

dashbd_indoorenv_sched
- grab classroom usage and reading data by joining schedule_min table with dashboard-with-indoor-envir
[used by dashbd-indoorenv-sched-testweathercombine]

dashbd-indoorenv-sched-testweathercombine
- grabs dashbd-indoorenv-sched and displays alongside weather data set to null
[used by dashbd_indoorenv_weatherstation]

dashbd_indoorenv_weatherstation
- union of dashbd-indoorenv-sched-testweathercombine and view_weather
[used by NA]

dashboard-with-indoor-envir
- union of kat."dashboard-union_individual" and view-indoor-envir
[used by dashbd_indoorenv_sched]

view-indoor-envir
- grab conditions of indoor environment
[used by dashboard-with-indoor-envir]

view_weather
- grab readings where building='weather-station'
[used by dashbd_indoorenv_weatherstation]

eileenpeppard, 2019-06-28:
Hi, Matt. Just wondering why you need descriptions.
-Eileen

matthew-schultz, 2019-06-28:
Hi Eileen, while I've been testing the old queries, I've also been copying them to sql files, in case they come in useful for future reference.

When talking about it with Carlos last week, I think the idea was that the descriptions might be useful for anyone referring to those files to understand what the queries do.

eileenpeppard, 2019-06-28:
Thanks, Matt.

matthew-schultz, 2019-07-03:
In the process of testing views it looks like I don't have permissions for tables __kevins_meter__ and __pmv_ip__
- When I attempted to run queries or the psql \copy command, they both returned errors like ```ERROR:  permission denied for relation kevins_meter```
- kevins_meter is used by the views meter-kevin-seconds and dates_kevins_meter (which are also used by other views)
- pmv_ip is used by the view dates-pmv

There are other views I should be able to test in the mean time, so I'll be working on those.

matthew-schultz, 2019-07-04:
Notes from talking with Eileen
---
- tables kevins_meter and pmv_ip (and the views dependent on them) are not used/not that important right now.
- readings with purpose_id's 98 and 111 are not used (see issue #96 for more info)
   - when testing views, we can filter out readings with these purposes

Notes from meeting with Carlos
---

View documentation is important; but don't spend too much time; testing that the views work properly is the main priority. If the views don't work, then the code is not useful for ERDL

Carlos and Eileen might modify the documentation in the future, if they have a clearer understanding of what views do.

Email Carlos latest branch running on sea-local but make sure branch is based on master

July 15th - check how far we are and decide when we are going to deploy to production
- contact/remind Carlos about deadline by then

Current priorities
---
1. queries
2. upload timestamp
   - handle duplicates
3. enum
4. rename locked_user to lonoa
5. verbose flag for logging
6. more code review (functions, naming, etc)
7. report module? or raspberry pi?

matthew-schultz, 2019-07-12:
@carlosparadis

Just wanted to note some differences that I found in readings while testing views, when I compared lonoa with scrape-util:

- purpose 82 (CO2)
   - gap on scrape-util in readings from 2019-06-14 13:30:00 TO 2019-06-23 07:40:00
   - not testing over gap for now
- purposes 13,14 seem to be swapped (duct-return-temp, duct-supply-temp in frog-1)
   - scrape_util and frog_uhm purpose_id to sensor_info.type mapping is not consistent/swapped
   - may need to edit either scrape-util config or their rows in sensor_info
   - excluding from tests for now
- purposes 98,111 (other-avg)
   - will be excluding these purposes' readings from testing, since they aren't used and seem to be calculated manually by scrape-util #96
-  purposes 118,119 (windows-open)
   - reading.units are different
      - scrape_util   "bool"
      - lonoa   "0 or 1"
   - not really a significant difference
- purposes 84,85 (egauge-power)
   - reading.units are different
      - scrape_util   "kw"
      - lonoa   "kW"
   - not really a significant difference

carlosparadis, 2019-07-12:
Great! 

- I think purpose 82 may have been a problem when the server belly up out of space. 

- Purpose id 13 and 14, I think the easiest option is simply to modify the **purpose** table there, and switch the purpose_id column of 13 and 14. Scrape-util does not care about the purpose table, and the readings do not need to be modified.  It will stay incorrect in the config file, but we are moving away from it anyway. **Have Eileen do this, so there is no confusion**. 

For purposes 98 and 111, ask Eileen if she or Kat can make a SQL Query in the database once we move on to Lonoa that does w/e scrape-util was doing. I would like to keep the code clean from transformations to avoid inconsistencies. It should just be a data acquisition pipeline as closer to raw as it goes. Not create new sensor columns that don't exist. If things needs get modified later, it will be stuck on code side, which is something we do not want. 

Make sure Eileen is aware of purposes id 118, 119, 84 and 85 remarks so it doesn't break anything on her Views or in Tableau.

eileenpeppard, 2019-07-12:
Hi, Guys.

Good job, catching the changes. I deleted some data for CO2 because the wiring on the sensor was messed up and giving bad readings. Yes, I swapped out information about two duct sensors since it was obvious they were confused on the original setup. Yes on the next two items, other was calculated and windows changed to 0 or 1. The correct way to express power units is kW. lower case, then upper case. Thanks! Great job!

matthew-schultz, 2019-07-12:
Hi Eileen,
Got it! Thank you for responding and getting that information to me so soon after my message

I can delete the same CO2 data on lonoa, so it doesn't include the bad readings.

If those duct sensors' purpose_ids and readings are already correct in the frog_uhm database, I might have to double check with you how I should update lonoa's sensor_info table in the test database, so it obtains the correct readings for those purposes.

I probably can do that for future readings by just swapping their query_strings in the sensor_info table in the test database.

Will also probably keep the "0 or 1" and "kW" units in lonoa if that sounds good.

matthew-schultz, 2019-07-16:
@carlosparadis 

Do we want to start planning to set up lonoa on sea-vm this week?

I was able to test all the public views, create documentation files for each of them, and classify them as egauge, webctrl, lonoa, or database views.

However, I skipped trying to compare __kat."dashboard-union_of_views"__ since I got an error trying to query that view on frog_uhm and after talking with Eileen it sounded like she uses the view kat."dashboard-union_individual" instead.

```
frog_uhm=> select * from kat."dashboard-union_of_views";
ERROR:  failed to find conversion function from unknown to character varying
```

---
Other notes

Views __single_line_CO2_temp_weather_ac_fans__ and __single_line_CO2_temp_weather_ac_runtime_fans__ needed to be queried on frog_uhm using quotes.

```
frog_uhm=> select * from "single_line_CO2_temp_weather_ac_fans" 
where datetime < (timestamp '2019-07-11 12:00') 
and datetime >= (timestamp '2019-07-11');
```

carlosparadis, 2019-07-16:
Ah I see, it is the "fated day". I gave you a Christma's wish list of to-dos and said we could revisit that list on July 15 to see how far you made it. Could you maybe give me a checkbox of the items and how far you went so I can make the call of moving to sea-vm? If we have a dedicated issue for said wishlist, reply to me on that issue instead and I will follow-up from there, so the discussion is not in the incorrect issue. I also want to know if update_datetime was added to the reading table on said issue or here if there is none.

Thanks!

matthew-schultz, 2019-07-16:
I didn't get too far down the list 😬

- [x] 1. queries
- [x] 2. upload_timestamp in reading table (implemented in [the branch](https://github.com/matthew-schultz/sensors/tree/94-update-tables-for-view-compatibility_ordered-columns) currently running on sea-local)
- [ ] 3. handle duplicates (I had this one under upload timestamp, but I'm not sure if we want to work on this one next)
- [ ] 4. enum
- [ ] 5. rename locked_user to lonoa
- [ ] 6. verbose flag for logging
- [ ] 7. more code review (functions, naming, etc)
- [ ] 8. report module? or raspberry pi?

matthew-schultz, 2019-07-17:
I'm going to see if I can get 4. enum update done tonight and maybe try to work on a few other items on the list.

carlosparadis, 2019-07-17:
@matthew-schultz You need to fix 5. before we can move to production to avoid creating an unnecessary user. 

I also don't want to end up having more than one version running in production and be forgotten accidentally, so try to stick with the git clone version of the production, and only pull from there, not from your fork or branches if possible.

We should also plan to clean the various versions from local and remove it entirely from there (no need to uninstall packages, just the actual code), but we should also remove locked_user from there to avoid long-term confusion.

carlosparadis, 2019-07-24:
Ok we should wrap this issue up. Make a Pull Request of the views for now on let's create folder `sql_views` on the top folder and put all files over there. Be careful with any sensitive information being put over there (mainly wants to look for anything like latitude and longitude). @eileenpeppard let me know if there is anything else.

matthew-schultz, 2019-07-31:
Closing this issue since PR #105 was accepted

closed 2019-07-31 by matthew-schultz