erdl/lonoa #97: Replace 'locked_user' with 'lonoa'

matthew-schultz, 2019-07-18:
@carlosparadis 

https://github.com/erdl/sensors/issues/94#issuecomment-512596049

I was able to create a branch based on erdl/sensors/master where 'locked_user' was replaced with 'lonoa'.

Should I go ahead and make a pull request with that branch?

Or do we want to use a branch rebased onto the latest branch running on sea-local? (has reading.last_updated_datetime and columns added/renamed to match scrape-util's sensor_info and reading tables)

carlosparadis, 2019-07-18:
Are there merge conflicts among the current pull requests? The common practice is to be based on production code.

matthew-schultz, 2019-07-18:
Sorry, yeah after doing a test rebase, I think my branch with all the updates conflicts with PR #81

Maybe I should just do a PR with the locked_user to lonoa renaming
and a PR with the reading.upload_timestamp update, if those are the most essential?

Also just want to make sure: by production, that would refer to the erdl/sensors master branch, right?

carlosparadis, 2019-07-18:
When I say production, I mean the server location in ITS, sea-vm. :) I will leave the choice of how to go about the branching up to you, so long it doesn't become a tangled mess. Refer to the videos from Udacity I refer to once about this if you're not too sure!

matthew-schultz, 2019-07-18:
Hmm, I think I'll just try to get in the 2 smaller pull requests for now, so we can check if they're good enough to run on production.

matthew-schultz, 2019-07-24:
Closing this issue since PR #98 was accepted

closed 2019-07-24 by matthew-schultz