erdl/lonoa #104: Migrate data from frog database to lonoa database

matthew-schultz, 2019-07-26:
- sensor_info
- reading
- etc.

matthew-schultz, 2019-08-08:
Notes from lonoa production setup today

- Copy readings before 8/7 from frog_uhm to lonoa_frogs
- Move any tables required by @eileenpeppard for views, etc, that are not already there.
   - __HIGH PRIORITY__ Create scripts to automate exporting (and importing) of those tables 
   - Complete import process after @kathrynparadis finishes checking over views

matthew-schultz, 2019-08-17:
Attempted to import these tables into lonoa_frogs first, since they are needed for the public views

- model_end_uses
- model_scenarios
- model_data
- schedule_min

They were migrated successfully but with the owner ```lonoa_frogs``` instead of ```sepgroup_nonmsg```

carlosparadis, 2019-08-17:
@eileenpeppard @matthew-schultz can you do a quick check the views hosted on Github folder and whichever other ones come to mind work now in production? 

@eileenpeppard Try starting using this database stead of the old one, there should not be anything left from your end use.

matthew-schultz, 2019-10-08:
closing since data has been migrated

closed 2019-10-08 by matthew-schultz