erdl/lonoa #127: Trade-offs between database column values and insertion of new sensors

carlosparadis, 2019-12-18:
This summarizes a point of concern about our meeting today for new sensors added in the future.

# Problem 

Currently, in the database schema created by `init_database.py`,  one column is created with enums: 

https://github.com/erdl/lonoa/blob/65609c55de1628484febae7d3730dd4f27456c26/init_database.py#L44

(@matthew-schultz I didn't realize you were using egauge orm here. Since we are moving this to a single file it's not too bad but if we didn't make in time it would cause confusion down the road on "why egauge and not the orm of another sensor".)

The database is created as part of a setup function leveraging sql alchemy library:

https://github.com/erdl/lonoa/blob/65609c55de1628484febae7d3730dd4f27456c26/egauge/script/orm_egauge.py#L171-L183

And in particular, the point of concern lies within this class of the data model in the same file:

https://github.com/erdl/lonoa/blob/65609c55de1628484febae7d3730dd4f27456c26/egauge/script/orm_egauge.py#L74-L80

1. According to this, the possible values of sensors are defined at the moment the database is **initialized**. Adding a new folder sensor, therefore requires not only code be added, but **manually** editing every database that exists on the server to add the new sensor. There is also no clear way that a new programmer will know they have to edit this column. 

2. There is also a concern if the responsibility of verifying, before adding to the crontab, if the script even exist. Currently, verification does not exist. `init_crontab.py` passively assumes, since the column sensor_type is enum, that the value will never be inserted incorrectly.  

# Solution

Not sure yet. The current implementation makes sense because values are inserted directly in the database, and an enum shows on Navicat as a dropdown menu instead of letting Eileen edit the text, which a typo would suffice to mess the crontab file. However, in the long-run due to 1) this solution may prove confusing to someone adding a new sensor or troublesome as they will have to manually edit the database. Code changes alone are insufficient. 

We would like a solution similar to what is observed on `init_crontab.py` on what concerns adding a new folder:

https://github.com/erdl/lonoa/blob/65609c55de1628484febae7d3730dd4f27456c26/init_crontab.py#L54-L62

So long a new script is added following the convention `api_` etc, modifying `init_crontab.py` is even unnecessary altogether!

matthew-schultz, 2019-12-22:
@carlosparadis 

Hmm it's difficult to find a more ideal solution for where to keep an authoritative list of  sensor_info.script_folder values.

After we combine the orm files into one orm file and moving to to the top level lonoa directory, the main choices I can think of that I could probably finish up soon are:

option 1.) Keeping the script_folder class in the orm file
   - Future updates would require:
      - updating the sensor_info.script_folder enum column in the database
      - updating the ScriptFolderEnum class in the orm file.

option 2.) Moving the ScriptFolderEnum class from orm file to init_crontab
   - update the orm file to import init_crontab.ScriptFolderEnum
   - Future updates would require:
      - updating sensor_info.script_folder enum colum in the database
      - updating the ScriptFolderEnum class in init_crontab

And then updating the wiki with a more clear section on adding a new sensor type section. 

I will try to update the wiki with the steps for the current method of adding new sensor types this weekend, unless we decide on a different option

carlosparadis, 2019-12-22:
@matthew-schultz Let me know what you think of this: 

In a package I use in R, called "pkgdown", there is a convention where any folder or file starting with **_** is to be ignored when performing package functions (e.g. generate docs). 

So if we were to use that here: 

 * egauge
 * _rhasberry_pi 
 * webctrl 
 * hobo
 * _weather_underground
 * _imonnit

Rhasberry Pi, Imonnit, and Weather Underground would both have a prefix of `_` because they don't currently have a /script/api_<sensor>.py, and hence Lonoa can't add data as it does to egauge-webctrl, hobo by having Eileen add it's sensor name to the **script_folder**. 

The _ convention is also used in programming languages to indicate its internal, or that it should not be expected to be messed with (e.g. C): https://stackoverflow.com/questions/39625352/why-do-some-functions-in-c-have-an-underscore-prefix 

I was thinking that if we capitalize on this to rename the folders, we can address the problem we currently have: That of the sensor names being hardcoded on the **orm.py**, and consequently old projects all have to be manually edited on the database manually to incorporate the new sensor, since **init_database.py** is only executed once.

**The solution** would simply have the **orm.py** enum class make the list of values it can take dynamical: We move the current sensor folders inside a folder called **sensors**, and adopt the _ notation. The orm then, at the time **init_database.py** is executed, checks lonoa/sensors for all folders which do not contain the preceding **_** and create the column. In the example above, that would be egauge, hobo and webctrl. 

We would, however, need to add something on `init_crontab.py` to alter the column **script_folder** every time it executes if it detects a new folder in there that doesn't match the available values. 

I don't know, at this point, if all this trouble warranties the dropdown menu on the column when Eileen is adding new sensors, or we shouldn't just turn it varchar.
Provided we explicit write on the README a "How to add a sensor" and mention people should not create a folder within sensors without a _, it should be fine. Eileen gets to keep a column where she can see the available sensor names, and

matthew-schultz, 2019-12-23:
@carlosparadis I like your solution!

That's pretty elegant how 
1) init_crontab would handle updating the values of the script_folder enum in the database 
2) the script_folder enum's values would be every named folder without a _ at the beginning.

Though we'd just have to see if it could be implemented and tested over the break.